olliw42/mLRS #1: Some code problem

Lance0724, 2022-01-18:
1. There are so many hard code local path(like "C:/Users/Olli/Documents/GitHub/.....") in the project file ( ".XXXXX" file  like .project/.cproject)
2. Implement the method of the class in the header file (like rxstats.h)
3. Define global variables in header files( in common.h , the section: "Common Variables "  )

Lance0724, 2022-01-18:
I`m The First ~ 😄

olliw42, 2022-01-19:
many thx

yeah, I'm not really happy with STM32CubeIDE ... and I use it only for short ... I noted these hard coded paths too, but I frankly have no idea how to overcome that 

any help of how to get STM32CubeIde to behave well such that it can be transferred would be massively appreciated

olliw42, 2022-01-20:
@Lance0724 
solved 1. by improving the setup procedure. Points 2. & 3. are really not code problems but rather taste. So I close the issue. Please do not hesitate to reopen if you feel it needs further consideration.

closed 2022-01-20 by olliw42

Lance0724, 2022-01-20:
[image]

Build Finished. 😄

olliw42, 2022-01-20:
great! many thx for the feedback