olliw42/mLRS #2: Engineering STM32CubeIDE 1.8.0 not smooth

whqsz, 2022-01-19:
Hello author, let us see hope on LORA mavlink. This open source project has been paying attention since last year. Recently, I found that there has been a major update on the progress. I will promote it in the hobby circle as soon as possible. Unfortunately, we strictly followed the Installation Bits and Bops steps and added the project to STM32CubeIDE 1.8.0. The project compilation was not smooth, and most of us did not understand how this IDE target works. So ask for help!
[image]

olliw42, 2022-01-19:
it seems you have not aimed at using the existing project, but to recreate it yourself from scratch

there is nothing bad about it, only that it requires that quite some many steps need to be done, in correct order, and correctly

could you try to use the provided project? In the other issue it was correctly noted that it is full of hardcoded folder names, but maybe we can work out something together, to get this toolchain working also for others.

btw, the documentation is really currently near zero, so pl never assume that such a thing like the "bit and bops" will work by just following the steps. The project is unfortunately not even near such a stage there even docu would be good. Still way too young. I kindly beg for your pardon and understanding.

All this would be areas there user inputs and help would be very much appreciated. I am currently really completely occupied with advancing the code to a first v0.1 version.

olliw42, 2022-01-19:
did some research, have not tried it, but maybe what one needs to do is not to just trying to open the project but to import the project

> How do I import a project into stm32cubeide?
> When the program has started, either click the [Import SW / TS project] button presented in the Information Center (refer to Figure 1) or navigate to the [File]>[Open Projects from File System...] menu command. 4. This opens the Import Projects from File System or Archive dialog box.18.04.2019

olliw42, 2022-01-20:
@whqsz 

I think the solution is simple: you need to change from Debug to Release

you can do that with the 4th icon in the top icons menu

when you do so you will notice that the three folders Common, CommonRx, modules in the Project Explorer on the left will move to the up.
the problem is that these files are not marked as linked for the Debug configuration and so not found by the compiler

hope that helps

olliw42, 2022-01-20:
@whqsz 
sorry, me again, I have improved the setup procedure, and also updated the description accordingly. Would be interested to know how it goes for you.

olliw42, 2022-01-20:
hey
@Lance0724 has tested the improved setup procedure and found it to work, see https://github.com/olliw42/mLRS/issues/1.
so I assume it also will for you and hence close this issue
please do not hesitate to reopen the issue if you feel it needs more attention, or raise a new issue in case of ongoing issues
many thx

closed 2022-01-20 by olliw42