olliw42/mLRS #4: No data for FM30 and DIY E20

whqsz, 2022-02-05:
I have been paying attention to the progress of the mlrs project, the hardware preparation is completed, but the radio configuration still does not understand the working principle.
I noticed that the author is testing the mlrs hardware on the opentx mavlink branch. Does it mean that the opentx main body is not supported?
Currently using edgex2.6 Chinese version, FM30 transmitter and DIY E28-F103C8T6 receiver, the hardware is working normally FM30 transmitter green light flashes, receiver 3 serial ports only have debug content, serial No. 1 is not detected SBUS signal. Does the FM30 hardware need additional modification?

olliw42, 2022-02-05:
hey

first off, please try to keep your repository up to date, I notice yours is 16 days behind, but really much is changing, including bugs here an there :)

second, please consider joining the rcgroups discussion thread. The github issues are commonly and normally not supposed to be used for support, but reporting specific issues with the firmware or feature requests, and alike. I think that's a very good practice. I have seen you on Discord, however I'm not really active on Discord, and also have my fundamental issues with it and similar platforms, so it's not my daily or main platform you would find me. :)

mLRS should work and is supposed to work with main OpenTx, and also EdgeTx, you don't have to use my MAVLink for OpenTx branch

as of today, the FM30 tx module can indeed not be used without hardware modification(s), and it indeed would need some more love (it was the first thing I used but then moved on to my own tx module...). These points are probably relevant:

- currently, the PPM-Pin1 is not supported in the firmware. This should not be so difficult to add, but I'm not sure it is worthwhile, since that way only the rc data could be send to the tx module, nothing else.

- for using the SPort-Pin5 one needs to modify the hardware of the FM30 tx module for it to work with the firmware. This does then allow using the SBUS or CRSF external module settings in the OpenTx transmitter. It would also be possible to change the firmware so this works, but myself I won't do it as there is so much else on the plate.

- the FM30's Bluetooth is not supported. That is, in order to gain access to the serial mavlink data, one currently has to desolder the BT chip and solder in two wires for Rx/Tx. Adding support for the FM30 bluetooth would be possible, ELRS seems to have done it, but I won't do it, i.e., someone else would have to add the code.

- the FM30's USB port is not yet supported. 

hope that answers some questions

btw, I am Olli, you can call me so :)

olliw42, 2022-02-05:
if you happen to have a second DIY E28-F103CB, you can use this as transmitter

olliw42, 2022-02-05:
these are the modifications I di to the FM30 tx module:

[image]

[image]

whqsz, 2022-02-05:
Thanks for the replies, I agree that this kind of question shouldn't be posted here, but the rcgroups discussion thread seems to limit my reply functionality :( For a long time. Yes I added your account to discord.
I have watched all the videos you posted on youtube, including the mlrs project, I can hardly find any instructions about FM30 hardware modification, I have tried to analyze the code to solve the problem but the ability is limited :), edgetx can't find the mavlink protocol option, I should How to do?

olliw42, 2022-02-05:
> I agree that this kind of question shouldn't be posted here, but the rcgroups discussion thread seems to limit my reply functionality

very strange
maybe you just want to contact a rcgroups admin, to sort this out, I'm sure they would try to be helpful

> I can hardly find any instructions about FM30 hardware modification

there is no docu for lot's of stuff still, that's how it is with fresh projects :) 
so just ask (ideally in the rcgroups thread ;) )

I just have added pics to the previous post of how to do the modifications

> edgetx can't find the mavlink protocol option

neither vanilla edgetx nor opentx do support mavlink, and hence do not have this option! 

if you want to use what I show in my MAVLink-for-OpenTx videos, you need to use the MAVLink-for-OpenTx firmware, not normal opentx nor edgetx

the mLRS project, however, as said before, CAN be used with any transmitter running plain opentx or edgetx. !!!

so, these two projects, "MAVLink for OpenTx" and "mLRS" should not be confused, they are two different projects !!

>  I should How to do?

at this point in time I see three options

1. you happen to have a second DIY E28-F103CB, you then could use this as transmitter module

2. you do at least the 5pin modification on the FM30 module. In the mLRS firmware you would then have to set 
#define SETUP_TX_SERIAL_DESTINATION     0
#define SETUP_TX_CHANNELS_SOURCE       2
in the common_conf.h file, and in OpenTx/EdgeTx choose the "SBus" option for the external module. 
If you also want the serial data, you would in addition have to do serial mod.

3. you do the 5pin modification on the FM30 module (only, the serial mod is not needed). In the mLRS firmware you would then have to set 
#define SETUP_TX_SERIAL_DESTINATION     1
#define SETUP_TX_CHANNELS_SOURCE       1
in the common_conf.h file, and you use the MAVLinkForOpenTx firmware on your transmitter, and choose "Mavlink" option for the external module.

closed 2022-02-06 by whqsz