olliw42/mLRS #11: SX1262, rx-main: can it be reset to Idle also when it is in receive state?

olliw42, 2022-02-28:
@Imen-D 

it would be great if you could test this:

it is about these code lines https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L729-L736

```
#ifdef DEVICE_HAS_SX127x
      if ((connect_state >= CONNECT_STATE_SYNC) || (link_state == LINK_STATE_RECEIVE)) {
#else
      if (connect_state >= CONNECT_STATE_SYNC) {
#endif
        sx.SetToIdle();
        sx2.SetToIdle();
      }
```

for the SX1276 I found that one needs to call SetToIdle() also when link_state is being switched to LINK_STATE_RECEIVE in the doPostReceive handler. I wonder if the code can be simplified, i.e., if the define can be removed and the same code be used also for the SX1262

I thus would kindly ask you if you could test it for your system if it would also work with this change

```
      if ((connect_state >= CONNECT_STATE_SYNC) || (link_state == LINK_STATE_RECEIVE)) {
        sx.SetToIdle();
        sx2.SetToIdle();
      }
```

Secondly, I wonder if calling SetToIdle() is needed at all for the SX1262, that is, if it would work if one would just remove (out-comment) these lines completely  (in this case one could define SetToIdle() to be just an empty function for the SX1262)

THX!

olliw42, 2022-02-28:
EDIT: the issue with the sx1276 showed up particularly when (1) the receiver was run for a long while with the transmitter powered down, and when powering up the transmitter (after 15 min or so). Also, the issue occured often when (2) powering up & down the transmitter repeatedly. Eventually the receiver wouldn't connect anymore. A repower of the receiver would "solve" the issue. Just so you know what to test for with the sx1262.

Imen-D, 2022-02-28:
I had test it, and this is working for sx1262. on the receive state, sx1262 will enter receive mode with zero timeout, so any command from host can be done. as with this code we command sx126x to enter FS mode.

olliw42, 2022-02-28:
first, many thx for taking the effort

so, to have that clear in my mind, you are confirming that the simplified code as in the above works for you

what is not so clear to me is, if it also works if these four lines are actually out-commented, deleted, i.e., if SetToIdle() could be simply an empty function?

Imen-D, 2022-02-28:
Sure, yes both of this code is working
[https://i.imgur.com/Nz5T1a5.png](url)
[https://i.imgur.com/h141hKC.png](url)

olliw42, 2022-02-28:
cool

MANY thx

so, when I completed my testing for the sx1280, I can make such a change

closing this issue now as it is answered. THX a lot

closed 2022-02-28 by olliw42

Imen-D, 2022-02-28:
you're welcome