olliw42/mLRS #12: PR "rx, that's the correct way of injecting radio_status" breaks things?

olliw42, 2022-03-02:
@Imen-D wrote here https://github.com/olliw42/mLRS/commit/92bcd2f914d1a90380ead24ed95fde947141a73e#commitcomment-67754991

> Hi olliw, with this commit I can't do write fast mission anymore, is this related ?

hmmm .... it all should work as before

can you pl tell me what "write fast mission" is? you mean, writing missions is not fast anymore, but very slow?

not sure you've noticed: the setup parameters have changed, you need to set now BOTH SETUP_RX_SERIAL_LINK_MODE and SETUP_RX_SEND_RADIO_STATUS to 1. Maybe it's just this?

olliw42, 2022-03-02:
ah, ok. I don't know about what this write fast thing is doing. I never actually really used missions, seems I need to learn something here :)

do you happen to know what what messages are used for that? 

would you have a mission which you could share with me to test?

> If I see the code, mavlink stream or message will hold on rx buffer until we have a complete and ok result ?

yes
this is needed for really doing it properly, also in what will come in future
this should not really add significant delay, because the baudrate of the uart is much higher than that of the link
maybe the issue is that "message buffer" is only one message, maybe one needs to add more

hmhmhm

olliw42, 2022-03-02:
great, thx, I'll try it this evenning

I can't however see yet why it would not work anymore ... will be interesting to see :)

olliw42, 2022-03-02:
@Imen-D 

could you test this changes to the code:

- for sending the messages, there is a protection with serial.tx_is_empty(). Remove them
https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mavlink_interface_rx.h#L96
https://github.com/DanielScocco/Simple-Huffman-Coding/blob/master/huffman.c
change to
 if (f_inject_radio_status) { ... }
if (f_msg_available) {}

- in addition increase the rx buffer for the serial uart to 1024, that is define UARTB_RXBUFSIZE 1024 in your hal file

I would be curious to know if this "resolves" the issue

THX

Imen-D, 2022-03-02:
I had test it but I'm sorry still can't do write fast. I have log in mission planner, not sure if this can help

// Normal write ///
INFO MissionPlanner.Controls.ProgressReporterDialogue - Focus ctl
INFO MissionPlanner.Controls.ProgressReporterDialogue - in focus invoke
INFO MissionPlanner.Controls.ProgressReporterDialogue - DoWork
INFO MissionPlanner.GCSViews.FlightPlanner - wps values 0
INFO MissionPlanner.GCSViews.FlightPlanner - cmd rows 158
INFO MissionPlanner.MAVLinkInterface - set giveComport  current False new True
INFO MissionPlanner.MAVLinkInterface - setWPTotal req MISSION_COUNT {"count":158,"target_system":1,"target_component":1,"mission_type":0}
INFO MissionPlanner.MAVLinkInterface - set giveComport  current True new False
INFO MissionPlanner.MAVLinkInterface - set giveComport  current False new True
INFO MissionPlanner.MAVLinkInterface - setWPint req MISSION_ITEM_INT {"param1":0.0,"param2":0.0,"param3":0.0,"param4":0.0,"x":-63356054,"y":1069490792,"z":59.75,"seq":0,"command":16,"target_system":1,"target_component":1,"frame":0,"current":0,"autocontinue":1,"mission_type":0}
INFO MissionPlanner.MAVLinkInterface - setWPint resp MISSION_REQUEST{"buffer":"/gT6AQEoAQD/vpG3","rxtime":"2022-03-02T13:42:54.2476856Z","header":254,"payloadlength":4,"incompat_flags":0,"compat_flags":0,"seq":250,"sysid":1,"compid":1,"msgid":40,"ismavlink2":false,"msgtypename":"MISSION_REQUEST","data":{"seq":1,"target_system":255,"target_component":190,"mission_type":0},"crc16":46993,"sig":null,"sigLinkid":0,"sigTimestamp":0,"Length":12}
INFO MissionPlanner.MAVLinkInterface - set giveComport  current True new False
INFO MissionPlanner.MAVLinkInterface - set giveComport  current False new True
INFO MissionPlanner.MAVLinkInterface - setWPint req MISSION_ITEM_INT {"param1":0.0,"param2":0.0,"param3":0.0,"param4":0.0,"x":-63343116,"y":1069464111,"z":100.0,"seq":1,"command":16,"target_system":1,"target_component":1,"frame":3,"current":0,"autocontinue":1,"mission_type":0}
INFO MissionPlanner.MAVLinkInterface - setWPint resp MISSION_REQUEST{"buffer":"/gT7AQEoAgD/vqDn","rxtime":"2022-03-02T13:42:54.4265499Z","header":254,"payloadlength":4,"incompat_flags":0,"compat_flags":0,"seq":251,"sysid":1,"compid":1,"msgid":40,"ismavlink2":false,"msgtypename":"MISSION_REQUEST","data":{"seq":2,"target_system":255,"target_component":190,"mission_type":0},"crc16":59296,"sig":null,"sigLinkid":0,"sigTimestamp":0,"Length":12}
INFO MissionPlanner.MAVLinkInterface - set giveComport  current True new False
INFO MissionPlanner.MAVLinkInterface - set giveComport  current False new True
INFO MissionPlanner.MAVLinkInterface - setWPint req MISSION_ITEM_INT {"param1":0.0,"param2":0.0,"param3":0.0,"param4":0.0,"x":-63339277,"y":1069470227,"z":100.0,"seq":2,"command":16,"target_system":1,"target_component":1,"frame":3,"current":0,"autocontinue":1,"mission_type":0}
INFO MissionPlanner.MAVLinkInterface - setWPint resp MISSION_REQUEST{"buffer":"/gT8AQEoAwD/vgYF","rxtime":"2022-03-02T13:42:54.6261278Z","header":254,"payloadlength":4,"incompat_flags":0,"compat_flags":0,"seq":252,"sysid":1,"compid":1,"msgid":40,"ismavlink2":false,"msgtypename":"MISSION_REQUEST","data":{"seq":3,"target_system":255,"target_component":190,"mission_type":0},"crc16":1286,"sig":null,"sigLinkid":0,"sigTimestamp":0,"Length":12}
INFO MissionPlanner.MAVLinkInterface - set giveComport  current True new False
INFO MissionPlanner.MAVLinkInterface - set giveComport  current False new True
INFO MissionPlanner.MAVLinkInterface - setWPint req MISSION_ITEM_INT {"param1":0.0,"param2":0.0,"param3":0.0,"param4":0.0,"x":-63337891,"y":1069485140,"z":100.0,"seq":3,"command":16,"target_system":1,"target_component":1,"frame":3,"current":0,"autocontinue":1,"mission_type":0}
INFO MissionPlanner.MAVLinkInterface - setWPint resp MISSION_REQUEST{"buffer":"/gT9AQEoBAD/vid4","rxtime":"2022-03-02T13:42:54.7967274Z","header":254,"payloadlength":4,"incompat_flags":0,"compat_flags":0,"seq":253,"sysid":1,"compid":1,"msgid":40,"ismavlink2":false,"msgtypename":"MISSION_REQUEST","data":{"seq":4,"target_system":255,"target_component":190,"mission_type":0},"crc16":30759,"sig":null,"sigLinkid":0,"sigTimestamp":0,"Length":12}

// Write Fast //
INFO MissionPlanner.Controls.ProgressReporterDialogue - RunBackgroundOperation
INFO MissionPlanner.Controls.ProgressReporterDialogue - Focus ctl
INFO MissionPlanner.Controls.ProgressReporterDialogue - in focus invoke
INFO MissionPlanner.Controls.ProgressReporterDialogue - DoWork
INFO MissionPlanner.MAVLinkInterface - set giveComport ProgressReporterDialogue Background thread current False new True
INFO MissionPlanner.MAVLinkInterface - setWPTotal req MISSION_COUNT {"count":158,"target_system":1,"target_component":1,"mission_type":0}
MISSION_REQUEST 0 {"seq":0,"target_system":255,"target_component":190,"mission_type":0}
INFO MissionPlanner.MAVLinkInterface - set giveComport  current True new False
INFO MissionPlanner.GCSViews.FlightPlanner - WP no 0 {"param1":0.0,"param2":0.0,"param3":0.0,"param4":0.0,"x":-63356054,"y":1069490792,"z":59.75,"seq":0,"command":16,"target_system":1,"target_component":1,"frame":3,"current":0,"autocontinue":1,"mission_type":0}
INFO MissionPlanner.GCSViews.FlightPlanner - WP no 1 {"param1":0.0,"param2":0.0,"param3":0.0,"param4":0.0,"x":-63343116,"y":1069464111,"z":100.0,"seq":1,"command":16,"target_system":1,"target_component":1,"frame":3,"current":0,"autocontinue":1,"mission_type":0}
[... cut ...] e":0}
MISSION_REQUEST 1 {"seq":1,"target_system":255,"target_component":190,"mission_type":0}
MISSION_ACK MAV_MISSION_INVALID_SEQUENCE {"target_system":255,"target_component":190,"type":13,"mission_type":0}
MISSION_ACK MAV_MISSION_INVALID_SEQUENCE {"target_system":255,"target_component":190,"type":13,"mission_type":0}
MISSION_ACK MAV_MISSION_INVALID_SEQUENCE {"target_system":255,"target_component":190,"type":13,"mission_type":0}
INFO MissionPlanner.Controls.ProgressReporterDialogue - DoWork Done

olliw42, 2022-03-02:
@Imen-D 
I'm trying to reproduce your issue
just to confirm we are seeing the same/are on the same page, can you tell me what error MissionPlanner gives you when you do a FastWrite?

Imen-D, 2022-03-02:
[image]

here

olliw42, 2022-03-02:
many thx
I do see the same :)

I also think I understood the issue now, one packet from the link can carry more than one mavlink message, and these are dropped with the current code ... just need to figure out what the correct algorithm would be :)

olliw42, 2022-03-02:
@Imen-D 
I just pushed a commit, which does solve it for me. Could you check if it also does for you?

MANY THX!!!
I really like a lot that you are doing all these in-depth tests ... I use only a small fraction of the MissionPlanner capabilities ... really great you do this

Imen-D, 2022-03-03:
Hi Olliw, thanks, working now

olliw42, 2022-03-03:
great
thx for testing and feedback
closing this thus

closed 2022-03-03 by olliw42