olliw42/mLRS #13: 31 Hz mode for SX1262 needs testing

olliw42, 2022-03-03:
@Imen-D 

I have just pushed support for the 31 Hz mode. On the Sx1280 it appears to work great. Could you test it also for the SX1262? All you need to do is setting the mode in common_config.h

THX in advance :)

Imen-D, 2022-03-03:
Sure, will let you know after test it

Imen-D, 2022-03-03:
Green led flash, amazing speed, its like we have rc radio and Sik radio on same H/W :)
Parameter download :
[image]

Write fast mission
[image]

Imen-D, 2022-03-03:
Very great olliw, I can confirm 31Hz working on sx1262

olliw42, 2022-03-03:
> Very great olliw, I can confirm 31Hz working on sx1262.

fantastic, MANY THX !!!

>  amazing speed, its like we have rc radio and Sik radio on same H/W :)

yeah, I can imagine that the 31Hz mode may become popular for those with sx1262

> You know what ?, I have sx1268 which is 433 band, I think can use same library with sx1262

well, this would be freaking cool 

I too believe that it should not be difficult to go to 433 MHz, probably just few settings to change (like LF instead of HF) ... and I also would expect that the sx1268 works like the sx1262 ...

the main problem with 433 MHz is that it is such a ....ing narrow band, at least here in Germany, and so heavily used...

olliw42, 2022-03-03:
anyway, main question answered, so closing

:):)

closed 2022-03-03 by olliw42