olliw42/mLRS #16: Prompt when compiling firmware: ld.exe: ./CommonTx/mlrs-tx.o:(.bss.gdisp+0x0): multiple definition of `gdisp';

suiaiyun, 2022-06-10:
https://github.com/olliw42/mLRS/blob/e0f8f91486ef010261720bb82fdebf697f5ff374/mLRS/CommonTx/gdisp.h#L110

It is suspected that the gdisp variable is created here. The gdisp variable is defined in gdisp C file, and in disp H file "extern tgdisplay gdisp;" Can be compiled and passed.

olliw42, 2022-06-12:
oh yeah, indeed
leftover from having had it h only initially 😂
thx for reporting
you want to raise a PR, or should I do?

suiaiyun, 2022-06-14:
> oh yeah, indeed leftover from having had it h only initially 😂 thx for reporting @suiaiyun you want to raise a PR, or should I do?

hi, olliw. I created a PR for issues #16. I wonder if it is appropriate for me to do so. This is my first time to submit PR in GitHub.

olliw42, 2022-06-14:
fantastic, many thanks, sir !!

yes, this worked perfectly. I have merged it. Note I did few formatting changes afterwards, but all nfc. 

Many thx again,
Olli

closed 2022-06-14 by olliw42

suiaiyun, 2022-06-14:
> 

OK, Olliw, I will try your method next time. I still know too little about GIT. Your method has benefited me a lot. Thank you again for your work. It's great.

olliw42, 2022-06-14:
>  I still know too little about GIT

yes, git can be really a pain for newcomers ... I went through this not too long ago so remember well 

>  I will try your method next time

choose whatever works best for you, I e.g. do nearly all with sourcetree nowadays, it was just a suggestion to give a pointer

> Thank you again for your work. It's great.

thx sir. For as long as you have fun with it, it has achieved all it's goals :)