olliw42/mLRS #18: Encryption options?

noahbliss, 2022-06-16:
I've seen AES and ChaCha as implementations in other radios, any plans for implementation here?

olliw42, 2022-06-16:
thx for bringing up the topic
I myself do not have plans in this direction, also because I have no expertise here
it would be however a useful addition, if it doesn't come at bandwidth costs
corresponding PRs I would sympathetically review :)

dizcza, 2024-04-30:
@noahbliss can you please write down examples of such radio packages that provide encryption?
I know only a very old and no-longer-maintained RadioHead.

marvp-afk, 2024-08-04:
There's another ELRS fork that incorporates encryption features: [PrivacyLRS](https://github.com/sensei-hacker/PrivacyLRS)

I'm personally lacking the skills to do this, but having "Privacy mLRS" would certainly be exciting, since it would be a much cheaper and open source alternative to the (afaik) only other option for encrypted MAVLink telemetry from RFD.
SiK's AES implementation is limited to the Si102x/Si103x, which are EOL and even if they weren't—still not LoRa.

closed 2024-09-03 by olliw42

sensei-hacker, 2024-11-10:
> There's another ELRS fork that incorporates encryption features: [PrivacyLRS](https://github.com/sensei-hacker/PrivacyLRS)
>
> I'm personally lacking the skills to do this, but having "Privacy mLRS" [...]

PrivacyLRS developer here.  In a couple months there may be a PR to allow high grade encryption with mLRS. 

Regarding @olliw42  's comment re bandwidth:
The encryption as currently done in PrivacyLRS incurs no additional bandwidth cost. It takes the same amount of time to 8 bytes that have been encrypted as it takes to send 8 bytes that have not.
Encryption provides confidentiality, privacy. By itself, it doesn't provide a strong guarantee that the message is legitimate - that it came from the desired control station.  The encryption by itself provides only a moderate degree of authentication. We'll also probably add an option for authentication, to have an absolute guarantee that the message came from the legitimate source. That authentication option would cost some bandwidth, when you turn that feature on.

Partly this depends on whether the MSP compliance / compatibility issue can be addressed first. It would need to work *reliably* with MSP first before we can then provide privacy via encryption, and authentication to ensure control messages really are coming from the legitimate control station. If it can be adjusted to follow the MSP protocol, then it can be reliable. After it's reliable we can add confidentiality and authentication.

olliw42, 2024-11-10:
I have to disappoint you. mLRS will not officially support encryption, and in fact discussion of this and other related topics is not welcome in mLRS channels.

The above post of me was made at a time when I haven't yet fully realized the implications of certain events.

I'll close this here.