olliw42/mLRS #19: add llcc68 chip support？

rsd1987, 2022-07-08:
Hi Olliw,
Semtech llcc68 is same to sx126x，only lacks some lower speed in lora mode，but is really cheaper than sx126x.

olliw42, 2023-03-07:
great
yo, I too had the impression it might be simply a drop in
but as always, the proof is in the taste of the pudding
will be interesting

jlpoltrack, 2023-03-07:
FYI, ELRS team found worse performance on SX126x and LLCC68 compared to SX1276 - thread here: https://forum.lora-developers.semtech.com/t/llcc68-poor-sensitivity/1480/14

olliw42, 2023-03-08:
interesting
thx for digging this out

I'm not sure I feel much discouraged by this however. I mean, it is certainly sad, and for those who want to reach the absolute limit no good news, but the sx1262 system otherwise offers so many advantages that - at least for me - it's still my prefereed choice in this frequency range. Among the advantages: 31 Hz mode, STM32WLE (= much simpler builds&designs) in form of modules, a suitable dev board. 

I myself also haven't seen the potential of 868/915 materialize so far, with both sx1262 based systems and the R9 system the link gets bad in my enviroment here at rssi's of about -100, not very different to what I see with 2.4 GHz, and far off any -110 or even -120 numbers ...

it's an important aspect to watch for sure however

jlpoltrack, 2023-03-20:
Quick notes on E220-900M30S module progress:
- Seems to use regular crystal instead of TCXO, added #define SX_USE_CRYSTALOSCILLATOR to Hal
- Can enter bind mode as an Rx but can't seem to bind to E5 Mini as Tx.  E5 CLI hangs when I enter bind mode

jlpoltrack, 2023-04-03:
@olliw42 
I think it's okay to close this.  Was able to get E220 w/ LLCC68 working using the SX126x driver without any changes.

olliw42, 2023-04-03:
so we can add LLCC68 as being supported to the main readme page?
that's cool
THX!!!

closed 2023-04-03 by olliw42

jebkerman-coder, 2023-06-25:
> @olliw42 I think it's okay to close this. Was able to get E220 w/ LLCC68 working using the SX126x driver without any changes.

Does this mean that the e22 module(sx1262) can be directly replaced with the e220 module(llcc68) without making any code changes? (They have the same pin definition)

jlpoltrack, 2023-06-25:
> > @olliw42 I think it's okay to close this. Was able to get E220 w/ LLCC68 working using the SX126x driver without any changes.
> 
> Does this mean that the e22 module(sx1262) can be directly replaced with the e220 module(llcc68) without making any code changes? (They have the same pin definition)

The only change needed for E220 modules is that you need to specify that they use crystal oscillators in the HAL (instead of TCXO).  This is a define and an example can be found here: https://github.com/olliw42/mLRS/blob/main/mLRS/Common/hal/tx-hal-E77-MBLKit-wle5cc.h#L103