olliw42/mLRS #24: STM32WLE5JC Support?

AlexKlimaj, 2022-09-28:
First of all, I love that this project exists. I have done some custom LoRa mavlink RTK correction links and have been wanting to do a generic LoRa based telemetry radio. I used the STM32WLE5JC with the built in LoRa radio and STCubeMX radio middleware.

I would be interested in producing hardware using this module and getting mLRS running on it.

https://www.seeedstudio.com/LoRa-E5-Wireless-Module-p-4745.html

olliw42, 2022-09-28:
yo, this Seeedstudio E5 module, and boards based on it like the E5 min dev board (https://www.seeedstudio.com/LoRa-E5-mini-STM32WLE5JC-p-4869.html), have been considered and discussed as very interesting options before, see e.g. the rcgroups thread. However, as always, some one needs to spend the time on doing the particular code for them.
I personally considered several times getting myself two of the E5 min dev boards, but honestly, I have spend already so much money on this project that I am very reluctant, and also it's really not like I have left over time.
So, to conclude: Yes, these boards are highly interesting for the project, and I would welcome any effort to get the code running on them. Any PR is welcome.

AlexKlimaj, 2022-09-28:
I have some of the dev boards I can send you. When I get some time, I could take a look at helping do the port as well.

olliw42, 2022-09-28:
> When I get some time, I could ...

well, you see, that holds true for me too :)
that is, your offer to send me some modules is really most generous, and much appreciated. However, I couldn't promise that I would have time to look at them anytime soon. So, pl, consider sending them only if you don't have any expectations with doing so.

I actually had looked a bit into the datasheet for the STM32WLE5, as well as into what CubeIde does, but it wasn't like I would have known immediately what I would have to do to port it to mLRS. It seems CubeIde only supports the SX with the HAL, but not LL, I would not want to see the HAL layer in mLRS, and I could see immediately if one could just use the mLRS code with minor adaptions or if there would be lots of work to do. What also has put me back a bit is that the errata sheet has quite some entries for the SX part, not sure if any is relevant, but that's not very encouraging. The bottom line, since you seem to have experience with coding for the STM32WLE5, significant progress would like only be made with some help/involvement of you

I just like to be frank about what the situation from my side is.

olliw42, 2022-09-29:
@ AlexKlimaj
so, just looked into the subghz HAL .h .c files ... it seems that it is pretty canonical ... it seems ther are all LL functions available though in the LL_RCC and LL_PWR sections ... I guess writing a wrapper to integrate with mLRS doesn't look very difficult.
if you would commit yourself to help with testing as well as would actually use it in real world then we could go trying that.

("using in real world" means this: I can't continously test it, simply for time reasons, so danger is that it gets bad when the mLRS code evolves, but since the mLRS code will evolve it means that there needs to be at least someone who routinely uses it so that introduced bugs are detected. That's why, for each feature which I can not routinely test myself, I want to have a "real world user which uses it in real world")(hope that make sense)

olliw42, 2022-10-04:
@AlexKlimaj 
you said you do want to do "hardware using this module". Do you have already some prototypes?

I mean, if you do, it might not make much sense to develop on e.g. the Seeedstudio Wio-E5 mini dev board, doesn't it

AlexKlimaj, 2022-10-04:
> @AlexKlimaj you said you do want to do "hardware using this module". Do you have already some prototypes?
> 
> I mean, if you do, it might not make much sense to develop on e.g. the Seeedstudio Wio-E5 mini dev board, doesn't it

Yes I can ship you a couple. It is a design I did for a customer. They have given me permission to ship them to you. 

Email me your address alex@arkelectron.com

olliw42, 2022-10-05:
@AlexKlimaj 
these would be also the ones you had in mind when writing "producing hardware using this module" ?

I want to ensure that I don't dev for the bin, i.e., I want to do it for a hardware which is or will be available to users. So either the thing you have in mind with "producing hardware using this module" or the Seeedstudio Wio-E5 mini dev board. But pl not some fancy thing which exists just in few hands. I hope it makes sense :)

AlexKlimaj, 2022-10-05:
It would be very similar. Same interfaces but with a JST-GH instead of whats on there now.

The Lora-E5 dev board breaks out UART1 to a UART to USB converter. My board bring out UART1 to the connector.
My board also has a Pixhawk standard debug port with SWD and USART2 broken out.

Lora-E5 schematic.
https://files.seeedstudio.com/wiki/Loramini/LoRa-E5%20mini%20v1.0.pdf

My board.
[image]

olliw42, 2022-10-05:
@AlexKlimaj 
many thx for sharing, very helpful

few comments, from a mLRS point of view
- it would be really helpful if the tx module would have a further uart port ... since from the scheme it looks it is not intended to be going into a JR bay of a radio, one will need CLI access. One could use uart2, but then one would lose the debug port.
- the CLI uart should be on the UART which allows firmware update, could be uart2 from the wle docs
- it would be really helpful to have a bind button, otherwise binding will be a user hazzle
- it would be really helpful to have two leds, a green and red, as it gives info the user I think may like
- it would be really helpful to have in/out pin, for crsf etc. pp

note that mLRS doesn't work like e.g. SiK radios. mLRS always will send also rc channels. Sure, you may not use them, but you would lose the bandwidth in this direction. Also, mLRS firmware for tx and rx is NOT symmetrical, unlike for SiK; they are different. As a consequence, also the hardware is asymmetric between tx and rx; the tx at least should have a way to access the cli or allow other method to configure the params

it looks to me that - at least for developing - the  Lora-E5 dev board is more convenient. I guess I should get me one.

a thing I have not yet understood about the STM32WLE ... does it have an internal reliable/stable oscillator? I mean, even ST recommends using the STM32WLE with a TCXO ... but there is it? I think we should use one to avoid any issues like those ELRS is experiencing.

AlexKlimaj, 2022-10-05:
I think the module only has 2 actual UARTs. Maybe bit bang GPIO as the debug/CLI third UART?

I was more thinking about this being a replacement for SiK telemetry radios (ground station to vehicle). From what you are saying, it is more a mix of Express LRS RC radio and Mavlink telemetry radio?

I like the idea of binding telemetry radios. Where SIK radios will just connect to any running radio.

I could send you a couple of Lora-E5 dev boards.

For a custom board design using the E5 module, would these be the requirements?
- JST-GH for serial and 5V power (UART1) - Main serial port
- JST-GH for serial and 5V power (UART2) - CLI serial port
- JST-GH bit bang serial port for debug?
- JST-GH for IO?
- Bind button
- Red and Green LEDs for user feedback

olliw42, 2022-10-06:
the debug in mLRS is only Tx, no Rx. And it can already be a software uart.
the device however also has the LPUART, which is fully functioning for our purposes

you can use mLRS as a replacement for SiK, that is just a serial on Tx side and a serial on Rx side. But this does not mean that it would work like SiK. The tx and rx firmwares are different, so when a device is flashed you can only use it as tx or rx, but can't swap it like with SiK.

well, the concept of a rc stream and bidirectional serial stream is 10 years older than ELRS ... but yes, mLRS combines both rc radio link and bidirectional telemetry link

the SiKs you don't have to "bind", but here you too need to set up the two partners correctly up, e.g. they need to have the same id, frequencies, etc.

"mandatory":
*  JST-GH for serial and 5V power (UART1) - Main serial port
*  ONLY on Tx: JST-GH for serial and 5V power (UART2) - CLI serial port.
*  Bind button
*  Red and Green LEDs for user feedback
* On Tx: In port, on Rx: out port, for crsf etc.. 

optional 
* JST-GH LPUART1 serial port for debug? For debug is very useful, but I use only Tx, not Rx. Could be also a soft uart, but since we have LPUART1 why not using it (when it is unused otherwise)
* SWD for flashing

very optional
* JST-GH for IO? This might be useful. For instance, one user wanted a buzzer/audio support. So, some few spare extra pin, chosen wisely, can be nice.
* On Tx: Yet another full uart (could be LPUART, and debug softuart). This would allow router configurations, E.g. one uart could go via BT/wifi to MissionPlanner, and the other uart to e.g. an antenna tracker.
* On Rx: May not hurt to have another uart available on Rx, may allow future things like a GPS? Could be what the CLI is for the Tx.

overall, you see, it is very flexible, and maybe overwhelmingly so at first. If you want the modules to be small, select fewer.

AlexKlimaj, 2022-10-06:
Hm, could the TX and RX mode just be set with a DIP switch? Reading the state of a GPIO on boot? So the same firmware can be used on both TX and RX?

I like that the radio pairs need to be bound. This is an improvement over SIK radios.

For the case of someone using a controller attached to a computer and the ground control station, the GCS is sending mavlink manual control messages through the telemetry radio. Would mLRS decode the mavlink manual control message and send it over its dedicated RC stream? Then the receiver on the vehicle would reassemble them back into mavlink messages before sending over serial?

I'd start with the Lora-E5 dev board. And I can work on a board with the exact interfaces and features.

olliw42, 2022-10-06:
> Hm, could the TX and RX mode just be set with a DIP switch? Reading the state of a GPIO on boot? So the same firmware can be used on both TX and RX?

well, technically yes, but I don't think this would ever make it into the main repo, as it really complicates the code enormously

> For the case of someone using a controller attached to a computer and the ground control station, the GCS is sending mavlink manual control messages through the telemetry radio. Would mLRS decode [...]

I just looked up MANUAL_CONTROL, and it is not clear how they should be translated into rc channels. If you can come up with an unambiguous scheme which translates the fields into the rc data fields of mLRS, one can consider this.

A similar thing is btw already done on the rx side: The rc data can be send out as RC_CHANNELS_OVERRIDE (or RADIO_RC_CHANNELS, a mLRS message I'm experimenting with).

Generally I am not a fan of such tweaks and am very hesitant, since it ends up with a zillion of parameters no one ever will understand anymore. So we will want some smarter way to do the configuration. Just adding more and more parameters isn't a route to future.

> I'd start with the Lora-E5 dev board.

yes, that's what I concluded too

AlexKlimaj, 2022-10-06:
I think RC_CHANNELS_OVERRIDE would be what I want. https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE

In the case of PX4, it treats those inputs the same as an RC receiver.

But it looks like QGC only sends the https://mavlink.io/en/messages/common.html#MANUAL_CONTROL message.

I think that could reasonably be converted into regular RC values.

olliw42, 2022-10-06:
> I think RC_CHANNELS_OVERRIDE would be what I want. https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE
In the case of PX4, it treats those inputs the same as an RC receiver.

same with ArduPilot

> But it looks like QGC only sends the https://mavlink.io/en/messages/common.html#MANUAL_CONTROL message.

interesting

just looked it up for ArduCopter, it seems to do essentially the same as with RC_CHANNELS_OVERRIDE
how is the situation with PX4?

> I think that could reasonably be converted into regular RC values. 

yes, looks so. I think we need to carefully compare PX4 and ArduPilot

I guess we however first want to wait until you have a setup to test & fly. But you should get the feeling that I rather like that idea :)

olliw42, 2022-10-06:
so I finally got around and triggered the button, and ordered me a Seeedstudio Wio-E5 min dev board (https://wiki.seeedstudio.com/LoRa_E5_mini/) and a Seeedstudio Grove - Wio-E5 (https://wiki.seeedstudio.com/Grove_LoRa_E5_New_Version/) (gasp)

the Wio-E5 is obviously of most interest as tx module, while the smaller size and now usb makes the Grove-Wio-E5 more interesting as rx module

would be good if you would have some of them around too, I would want to rely on you concerning testing the code :)

olliw42, 2022-10-07:
just as a follow up of the above RC_CHANNELS_OVERRIDE/[MANUAL_CONTROL](https://mavlink.io/en/messages/common.html#MANUAL_CONTROL) substitution discussion, a similar discussion occured before also in the rcgroups thread
https://www.rcgroups.com/forums/showthread.php?4037943-mLRS-Lora-based-Mavlink-oriented-open-source-radio-link/page12&perpage=50#post49477527

vrquaeler, 2022-10-09:
Hi Olli, 
Haven't seen this discussion earlier. Happy to see you pulled the trigger. 
The chance of running on "off the shelf" hardware improves the accessibility of the mLRS project.

> so I finally got around and triggered the button, and ordered me a Seeedstudio Wio-E5 min dev board (https://wiki.seeedstudio.com/LoRa_E5_mini/) and a Seeedstudio Grove - Wio-E5 (https://wiki.seeedstudio.com/Grove_LoRa_E5_New_Version/) (gasp)
> 
The "New_Version" is related to Seeed's predecessor. The earlier model is based on a RFM95 module and an Atmega168. The Atmega168 handles the SPI to UART translation.
(https://www.seeedstudio.com/Grove-LoRa-Radio-868MHz.html)

Looking forward to do more testing.

olliw42, 2022-10-11:
Hey @AlexKlimaj 
I got my Wio-E5 dev boards few days back, and I made some progress in that I just have achieved a successful connection between the Wio-E5 mini dev board as transmitter and an E22-based receiver module. There are still some things to check and work on, but with a connection established it should be easier now to move forward.

My plan is to provide a target for the Wio-E5 mini dev board as transmitter and for the Grove-Wio-E5 as receiver. My question to you would be: Would this be a configuration which you could test and use (i.e. you have one of each boards)?

olliw42, 2022-10-12:
support for STM32WLE and for the two Seedstudio boards mentioned is now in mLRS.

I thus close this issue.

For further discussion on board design and so on, please do not hesitate to join the rcgropups thread, the discord server, or to open a new issue.

closed 2022-10-12 by olliw42