olliw42/mLRS #34: Bug: No characters allowed in Pass Phrase via CLI.

vrquaeler, 2022-11-01:
Assuming I am not totally wrong with the CLI.
While testing different targets I found a minor issue: 

Changing the Pass Phrase via CLI.
I am not allowed to enter character's in the pass phrase via CLI. 

**Observed  behavior:**
Enter: `p bind_phrase = wle5jc;`
Response: `err: invalid value`

**Expected behavior:**
Enter: `p bind_phrase = wle5jc;`
Expected response: `Bind Phrase = wle5jc`

Entering digit's or `#-._` works as it should. For example:
Enter: `p bind_phrase = 123456;`
gives the expected response: `Bind Phrase = 123456`

**Way to reproduce the effect:**
see "Observed behavior". 
Tested on _tx-R9M_, _tx-R9MX_, _tx-WIO-E5-mini._

**Additional info:** 
On the target _tx-R9M_ the Pass Phrase can be changed (incl. characters) and saved as it should.
The tests are done on Win10 and Linux PC's using the language setting: US-English UTF-8.

olliw42, 2022-11-01:
@vrquaeler 
oh, indeed MANY THX!!
this was actually an interesting one LOL
pl try out the latest, v0.3.05, should all be working now

vrquaeler, 2022-11-01:
@olliw42 

Have tried v0.3.05 ...
Everything is working as it should. Now running the CLI i can do everything I want to do. :+1: :+1: 
Thanks for the fast fix.

closed 2022-11-01 by vrquaeler

olliw42, 2022-11-01:
many thx for the feedback
great to hear