olliw42/mLRS #74: LUA support small screen ?

jasonwang01, 2023-04-27:
Is it possible that LUA support some radios with internal RFs and small screen(128x64)?  It only can use CLI to change param with PC.

olliw42, 2023-04-27:
not currently - unfortunately.

I do not have such a radio so I cannot do the lua script, so we would need someone with such a radio and the skils and wills to modify the lua script to work on the smaller displays.

A respective PR would be very much appreciated.

It's indeed one of the bigger things still missing.

vrquaeler, 2023-04-28:
At least I have the hardware available for testing. 
My LUA skills are just good enough for "one-line-code-changes".

olliw42, 2023-04-29:
@vrquaeler 
well, I have never looked into what needs to be done for the smaller screens, but I guess all it might be is to just adjust the layout, i.e., the x,y coordinates in the print commands, and of course color.
Could be that your "one-line-code-changes" skills are way good enough for that task :):)

vrquaeler, 2023-05-01:
It's on my list.

Maybe for the rainy days...

4O6FPV, 2023-05-15:
hmmm I have the radio and will soon be one handed from shoulder surgery so I may take this up!!!!

jasonwang01, 2023-05-22:
> not currently - unfortunately.
>
> I do not have such a radio so I cannot do the lua script, so we would need someone with such a radio and the skils and wills to modify the lua script to work on the [...]

Hello olliw, I had worked in lua with small screen radio at last weekend.
It only have MCU internal RAM, so free ram of opentx is too little for lua.
For example, DEVICE_PARAM_LIST is too large, radio report not enough memory, and some lua api like table.insert is not supported.
I plan don't storing all parameters, so a cmd like MBRIDGE_CMD_PARAM_REQUEST_BY_INDEX is needed.

olliw42, 2023-05-22:
@jasonwang01 
first, MANY thx for jumping in and looking at this. Massively appreciated :)

argh ... I was suspecting that it may not be that easy ... very sad

btw, do we know how the TBS configurator is doing it (which AFAIK is also used by ELRS)? it could maybe give us ideas

so, you want to read the list and plot the options with the current setting while reading all options, and want to keep the list of possible settings only for the presently selected option, right? So you need a command to retrieve this list for said option?

I think the precursor is already there, though not yet implemented, see https://github.com/olliw42/mLRS/blob/dev-mavlinkx/mLRS/CommonTx/mbridge_interface.h#L471C34-L477. Instead of defining a specific cmd, the approach would be to send a MBRIDGE_CMD_REQUEST_CMD with the cmd MBRIDGE_CMD_PARAM_ITEM, and the param index as further parameter. 
So, it's a bit more complicated than what you suggest, but not very much. This line https://github.com/olliw42/mLRS/blob/main/lua/mLRS.lua#L463 would be an example of how that would work with sending a cmd request.

So, we need to do the implementation side. Should not be very difficult, but needs to be done. I've raised an issue to put this in the list :). https://github.com/olliw42/mLRS/issues/82

zanefond, 2023-06-12:
hello ！ I have tested the black-white screen lua on my x9d+with edgetx 2.8.there are some problems while using the luascript on x9d+.first,when starting the luascript, the screen will showing mlrs loading,but cannot active the function. when changed the baudrate from 115K to 400K,I could enter the main page but still cannot calibrate the parameters.

olliw42, 2023-06-13:
> where can I get the lua for a x9d?

here is jasonwang01's PR: https://github.com/olliw42/mLRS/pull/83

geofrancis, 2023-07-01:
The script works great on my Taranis X9E with opentx.

olliw42, 2023-07-01:
many thx for the report !

@vrquaeler @geofrancis 
the question is also a bit how usefull it is. In my understanding it so far only allows to adjust 5 settings, not all. In your opinion, is that usefull enough for a good number of users to be happy with it and benefit from it? 
(question essentially is if it is ready enough to be spread to the public, or if it rather should see more improvement before)

geofrancis, 2023-07-01:
It does the essentials, the only other option I would like to see would be to receiver power level as well as transmitter power out. Other than that it's good enough.

olliw42, 2023-07-01:
thx!
so what options does it actually allow to set?

from the PR description and the code I deduce 
- BindPhrase
- Mode
- TxPwr
- RxPwr
- RxOutMode

the receiver power level should thus actually be available !?
(comment: the rx needs to be connected to the tx for the rx parameters to be available&settable)

vrquaeler, 2023-07-01:
Hi,

Even with the limitation to the 5 parameters mentioned above It is really useful as it is now. Otherwise we would have nothing for the BW-radios and are forced to use the CLI any time.  

The CLI has to be used for the other parameters. 

For parameter changes in the field like changing power level for a range check (without the need of a PC) I am fine with.

olliw42, 2023-07-01:
many thx

you find that it's possible to set also rx power?

vrquaeler, 2023-07-01:
Yes, it's possible to change the RX power. 
The RX has to be connected. 

Tested on Seeed E5 based hardware.

olliw42, 2023-07-01:
MANY THX for having tested it, and MANY THX to @jasonwang01 to have made that bw lua script.
I have merged it now.
I also opened a new issue about improving it to more parameters.

closed 2023-07-01 by olliw42

geofrancis, 2023-07-01:
mine displays the power level for both tx and rx but i can only change the tx power level as far as I can see.