olliw42/mLRS #81: SX128x OTA times don't match calculator?

jlpoltrack, 2023-05-19:
SX128x | 50 Hz shows OTA of 7892 uS but calculator shows 8128?

https://github.com/olliw42/mLRS/blob/main/mLRS/Common/sx-drivers/sx128x_driver.h#L57

[image]

SX126x matches calculator, e.g. 31 Hz:

https://github.com/olliw42/mLRS/blob/main/mLRS/Common/sx-drivers/sx126x_driver.h#L43

[image]

jlpoltrack, 2023-05-19:
Answering myself, mLRS uses the 'LI' Long Interleaving Coding Rates which align:

[image]

closed 2023-05-19 by jlpoltrack

olliw42, 2023-05-19:
exactly!
it's the reason it uses LI. I actually wonder if not using LI would give some benefit with the crc1 mechanism, but for now it's as it is :)