olliw42/mLRS #88: Dropping the "match to spektrum" rc code

olliw42, 2023-07-01:
There is currently still a piece of code which makes a spektrum systems and mLRS simklar as regards the rc signals: https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/out.cpp#L131-L138

That's great for all spektrum users, but not for many others. I agree that this piece of code should go, at some point. And I'm not against to just do it "now".

However, when removed, it will break everyone's installation. It thus would be good to spend few seconds of thought beforehand about when to do it and how to orchestrate it. Is there a better point in time than "now"? Is it sufficient to announce it in the annoucement discord channel? What else should we do? Or should we just do it and see what happens?

(@jlpoltrack)

Target0815, 2023-07-01:
It is IMHO better to make this change now than when mLRS is really round and many users use the system. 

As for Spektrum, all I can say is that I don't know anyone who uses the system for Mavlink and/or Long Range. The vast majority probably has a transmitter with OpenTX or EdgeTX or maybe Ethos.

olliw42, 2023-07-01:
many thx

> It is IMHO better to make this change now than when mLRS is really round and many users use the system.

I tend to agree :)

> As for Spektrum, all I can say is that I don't know anyone who uses the system for Mavlink and/or Long Range. The vast majority probably has a transmitter with OpenTX or EdgeTX or maybe Ethos.

I sure would no argue that spektrum is very relevant, but I do argue that conversely TBS/ELRS are not the only relevant systems.

Note, the "issue" is not really OpenTx/EdgeTx or not, but what the link system is, ie Frsky/Crossfire/Tracer/Ghost/Futaba/Spektrum/Flysky/... ... ...  from what I know from some friends you can't even rely on that FrsKy are drop-ins ...

vrquaeler, 2023-07-01:
Hi Olli, 
I have read all the comments (found in various stages of "design_decissions.h")
and am aware of the effort you already spend in servo scaling for different RC protocols.

IMHO any other additional parameter to mimic different RC systems would unnecessary bloat the software. We should avoid this.

Get rid of the re-scaling is the cleanest way software wise and the belonging maintenance. 
It would be my preference. 

Similar to Target0815 thoughts: The earlier the better.

olliw42, 2023-07-01:
many thx too!

> Get rid of the re-scaling ... The earlier the better.

copied :)

jlpoltrack, 2023-07-01:
No concerns from me with removing.

olliw42, 2023-07-03:
done with commit https://github.com/olliw42/mLRS/commit/a70e9fe887e92df8beedc0f70fb48949d5911034

closing

MANY THX for the feedback!

closed 2023-07-03 by olliw42