olliw42/mLRS #89: Lua for BW screens: Support more than 5 parameters

olliw42, 2023-07-01:
follow up on https://github.com/olliw42/mLRS/issues/74.

The current lua script for bw screens supports only 5 parameters, for reasons of limited ram on radios with bw screen.

It is however possible to read parameters also one by one, by using the parameter-by-index command, which should allow extended functionality on bw screen radios.

geofrancis, 2023-07-18:
I noticed that when i had the wrong firmware it would display the wrong parameters, that suggests it can display any parameter but only 5 at once. would it be possible for the user to select the index numbers for the parameters they want access to? so it would be possible to select what 5 parameters it displays.

olliw42, 2023-07-18:
>  would it be possible for the user to select the index numbers for the parameters they want access to? so it would be possible to select what 5 parameters it displays.

the 5 indices are defined here: https://github.com/olliw42/mLRS/blob/main/lua/mLRS-bw.lua#L27-L31
so it might be that all you need to do is to change them as you like
just a guess though

> also im struggling to get it to connect reliably, is there any prerequisites for using the lua. ? I generally find I have to restart everything a couple if times for it to work.

pl note that we do not appear to have any maintainer for that bw lua script. This also implies that it may function worse and worse over time when mLRS progresses. I probably was not a wise decisssion to "officially" include it.

zanefond, 2023-08-05:
the new lua script works well on my frsky x9d+ （ edgetx sys and  resistance modified ）

galaxyrepairman, 2023-11-28:
Hello, may I ask which signal line is used as the output of CRSF for E77 MBLkit TX? Is the E77 MBLkit successfully aligned? LED 3 will light up

brad112358, 2024-11-02:
Just a heads-up that I have been working on improving the BW lua script.  So far, I have a crude version which allows most parameters to be edited via several pages where only a page (7 parameters) are loaded at a time via MBRIDGE_CMD_PARAM_ITEM which is faster than the current method.  Unfortunately, I'm currently having problems with obtaining 3 specific parameters which somehow don't seem to have MBRIDGE_CMD_PARAM_ITEM2 processed before ITEM3 is encountered.

olliw42, 2024-11-02:
i.e., you are using the parameter-by-index command as in the one example lua?

I want to modify also the color lua such as to use that mechanism ... should get rid of the relod ... so your work here will hopefully be usefull also there :)

brad112358, 2024-11-02:
Yes, like the example, except just requesting the parameters on the current page instead of all of them sequentially.

I'm not sure what the trouble is with the ITEM2 which seems to be missing.  Do you know how large the queue is for crossfireTelemetryPop()?  Surely it must be enough for the 3 or 4 packets I'm expecting since I request one parameter at a time.  If not, perhaps slowing down would indeed help.

olliw42, 2024-11-02:
it actually may be just one packet, isn't it?

PR #229 (2024-11-15): Add multiple page support to bw lua to support all parameters
Some versions of this have been tested on X9D and X9D plus by other users.
Also added README.md with workaround in case of "not enough memory" error.
resolves #89
closed 2024-11-15 by olliw42