olliw42/mLRS #95: Main branch fails to compile?

jlpoltrack, 2023-07-13:
Tried to build and compile from main today:
[image]

Got a bunch of errors when compiling around ST drivers missing.  (I followed the usual steps with running 'run_setup.py' and had the correct workspace in STM32CubeIDE)

Had to use this commit in order to compile:

[image]

olliw42, 2023-07-13:
ok

I indeed made quite substantial changes to the "build system". The main driving force is to have an aoutomated way to set compile flags.

it also gets rid of a thing which really started to massivly annoy me, namely that all the ST drivers were stored in each target (which blew up the download but also required massive comits)

can you please go to the tools folder and run run_copy_st_drivers.py. This should solve the issue, hopefully. I think the bug is that this now needs to be also called from setup.py

jlpoltrack, 2023-07-13:
That script fails to run for me 'The system cannot find the file specified'

[image]

I do see files in that location: 
[image]

olliw42, 2023-07-13:
this is not windows, right?

jlpoltrack, 2023-07-13:
Windows 11

olliw42, 2023-07-13:
then I'm surprised...
I was thinking that the issue are these lines https://github.com/olliw42/mLRS/blob/main/tools/run_copy_st_drivers.py#L25-L26, but they work for me, on win ...

do you maybe have to call it with python3?

jlpoltrack, 2023-07-13:
Silly me, I had a space in my folder name 'mLRS Dev' which this script didn't like, changing to 'mLRSDev' works.

olliw42, 2023-07-13:
great!

and not so great ... it should work with blanks ... I will check what I find on this

but it needs to be called also from run_setup, that's clearly a bug you uncovered :D

olliw42, 2023-07-13:
it now should work

THX for discovering these issues!

jlpoltrack, 2023-07-13:
Perfect, latest commit resolves this issue.

closed 2023-07-13 by jlpoltrack

olliw42, 2023-07-13:
it should also work with blanks now.

THX, this was very helpful.

jlpoltrack, 2023-07-13:
Just confirming that it works with blanks/spaces too.