olliw42/mLRS #129: prep for v1.0

olliw42, 2023-09-27:
@jlpoltrack @brad112358 

I'm considering releasing main as v1.0.

There was one thing I wanted to do, however, which I have just pushed. The change affects the behavior of the sx handling. So far, the philosophy was kind of "in case of issue stop device with error indication" (aka FAILALWAYS), which however is not appropriate for a production release version. Here we rather want the system to try to recover itself. 

Anyway, it's a change in a "mission critical" part, so I would appreciate if you could give current main some testing.

@brad112358: do you see any blockers we should consider befor a v1.0 release? (@jlpoltrack has answered that question alreaady :))

jlpoltrack, 2023-09-28:
Can at least share that have done 1 flight on 31 Hz LoRa and 1 flight on 111 Hz FLRC - nothing out of the ordinary to report.

olliw42, 2023-09-29:
👍 👍 👍

closed 2023-09-30 by olliw42

brad112358, 2023-10-08:
@olliw42, I'm sorry, I'm so late responding to this!  I was quite busy with some traveling.  Congratulations on this milestone!  I had hoped to find time to work on scripting RX firmware update by now.  Hopefully, I'll get to that soon.  This also seems like a good time to get mLRS into the Ardupilot documentation wiki so I'll look into that...

olliw42, 2023-10-08:
hey @brad112358 
no problem :)
many thx, I'm glad too I followed your suggestion to put out a v1.0. Let's see how it goes.
I'm not sure chances are high to get it into the AP docs, given there are no available ready to use hardware, but I hopefully am wrong. I would think a blog post in the AP discuss forum could be a great alternative.
Most important however, it seems to work well for some users satisfyig them.