olliw42/mLRS #139: R9 MX (+ similar) only support 13 dBm?

jlpoltrack, 2023-11-27:
Recently acquired an R9 MX and the Lua returns 13 dBm when choosing the 17 dBm option.

Looking at the code I see a few things I don't understand:

1. Max Power is limited to 15 dBm even though 17 dBm is allowed for continuous use in the datasheet: https://github.com/olliw42/sx12xx-lib/blob/main/src/sx127x.h#L253, https://github.com/olliw42/sx12xx-lib/blob/main/src/sx127x.h#L261
2. The 15 dBm limit is applied when 17 dBm is asked for here: https://github.com/olliw42/mLRS/blob/main/mLRS/Common/sx-drivers/sx127x_driver.h#L49
3. There's a 2 dBm negative offset applied that I don't understand: https://github.com/olliw42/mLRS/blob/main/mLRS/Common/sx-drivers/sx127x_driver.h#L53

So it seems that when using a 'bare' SX1276, when 17 dBm is asked for you only get 15 dBm but the Lua returns 13 dBm due to the offset.

Should 17 dBm be allowed?  Where does this 2 dBm offset come from?

olliw42, 2023-11-27:
I can't tell you what's going on from scratch, but tell that this equation is at the bottom of it all: https://github.com/olliw42/mLRS/blob/main/mLRS/Common/sx-drivers/sx127x_driver.h#L45

jlpoltrack, 2023-11-27:
> did I maybe got the signs for the 2's wrong?

I think this is it.  For Pout = 17 dBm, outputPower should be set to 15.  17 - (15 - 15) -> 17 - 0 = 17

Current code adds 2 when it should be subtracted.

olliw42, 2023-11-27:
yes, looks like it is wrong in the two places

closed 2023-11-28 by olliw42