olliw42/mLRS #143: Compilation for dualband 868/915MHz & 2.4GHz

norulers, 2023-12-11:
I did not find the tx-hal-easysolder-e77-e28-dualband target in the dev-dualband2 branch, is it using the same target compilation as tx-hal-easysolder-e77-e22?

olliw42, 2023-12-11:
yes. There is no `tx-hal-easysolder-e77-e28-dualband` target but for laziness the branch simply reuses the `tx-easysolder-E77-E22-dual-wle5cc` target. 
(note, this could be what you might do for your own custom targets too, https://github.com/olliw42/mLRS-docu/blob/master/docs/CUSTOM_BOARDS_TARGETS.md#creating-a-target-for-a-custom-board)

allow me some comments to maybe help you steer through the code a bit better
A device's related info is found in `device_conf.h`, `hal.h`, and the respective hal file like `tx-hal-xxxx.h`.
For each different target is associated a define like `TX_DIY_E77_E22_WLE5CC`. This define is set in the target's properties tab. Usually however the target's name and this define's name are pretty similar, in the sense that from common sense they usually can be uniquely correlated.
So, in your case you would go to the `device_conf.h`, find the section https://github.com/olliw42/mLRS/blob/dev-dualband2/mLRS/Common/hal/device_conf.h#L302C8-L337, and from that it is enclosed by a `#ifdef TX_DIY_E77_E22_WLE5CC` determine that the target could be `tx-easysolder-E77-E22-dual-wle5cc`. In order to be absolutely sure you would right-click on the openend target and go to Properties->C/C++ Build->Settings->MCU G++ Compiler->Preprocessor and find `TX_DIY_E77_E22_WLE5CC` defined.

Note that the dual band branch has gotten a bit outdated.

norulers, 2023-12-11:
Thank you for your reply, I noticed the #if 1 #if 0 in the device_conf.h file, and the #if 0 part is compiled by default. I don’t know where to define 1. Will the e77-e28-dualband target merge?

olliw42, 2023-12-11:
constructs like `#if 1` are just another way to comment in/out code pieces

> Will the e77-e28-dualband target merge?

can't say. If some user(s) uses it regularly, so that potentially still remaining little issues can be weeded out and the code part is regularly tested, then yes. Otherwise not so likely.

olliw42, 2023-12-14:
@norulers for your info, I've updated the dev-dualbenad2 branch to sync with main

norulers, 2023-12-14:
> @norulers for your info, I've updated the dev-dualbenad2 branch to sync with main

Thank you for your reply, I will try the latest branch.

norulers, 2023-12-19:
Is the maximum power setting in the script for dualband 868/915MHz & 2.4GHz only 158mw correct? How to set the 500mw power of E28?

olliw42, 2023-12-19:
good question
currently there is no option to choose the powers for the E22 and E28 independently
currently they are simply set to equal values, see here: https://github.com/olliw42/mLRS/blob/dev-dualband2/mLRS/Common/setup.h#L503-L509
this is certainly a point which can be improved

for the E22/E28 combo it could be meaningfull to simply scale the E28's power up by 5dB