olliw42/mLRS #144: rx-R9MM: only sbus output

twistedwings, 2023-12-19:
rx-R9MM-f103rb-elrs-bl-v1.1.01-@19066cb.elrs

receiver outputs sbus instead od crsf.

mLRS.lua: Edit Rx: Out Mode is fixed to sbus (disabled).

olliw42, 2023-12-19:
R9MM's hardware allows to only do sbus. See also docs: https://github.com/olliw42/mLRS-docu/blob/master/docs/FRSKY_R9.md#r9-mm--r9-mini-receiver

closing since no issue :)

closed 2023-12-19 by olliw42

arg7, 2024-01-19:
Sorry, does it means that R9MM will not be able to transmit mavlink telemetry? 

Since mLRS Rx Module Setup tells to set  Rx Out Mode = crsf.

I'm trying to setup mLRS on R9M TX and R9MM RX and getting no output on R9M TX serial port (checked with osciloscope, and left dip switch is up).

Thanks,
Andrey

twistedwings, 2024-01-19:
no, it means that RC output is sbus only, no crsf.
Mentioned in the doc, overlooked that as I created the issue.
Works quite well.

olliw42, 2024-01-19:
as @twistedwings already said, you get both, the rc data and the mavlink data. All it means is that you do not get the rc data as CRSF but as sbus. But you still get the mavlink.
In the wiring diagram in the docs you see three wires: two green wires for the (mavlink) and one yellow wire for the rc (=sbus) data.
Pl ensure that your wiring is ok, and also ensure that you have done all teh settings as described in the docs, e.g., the "[CRSF Telemetry and Yaapu Telemetry App](https://github.com/olliw42/mLRS-docu/blob/master/docs/CRSF.md)" doc if this is what you want

arg7, 2024-01-19:
Thanks for prompt reply, I have double checked wiring to the RX, I see mavlink messages incoming on CH4 of R9MM and outcoming messages on CH3.

In mission planner I see that radio transmission works, in radio calibration panel gauges are moving.

But on the radio (RadioMaster TX16S, OpenEdge 2.8.4), the serial line of R9M TX module is silent and Yaapoo widget shows "no telemetry data".

What could be wrong with my setup?

Thanks in advance,
Andrey

olliw42, 2024-01-19:
my first guess would be that you may have to upgrade EdgeTx, 2.8.4 is pretty old

does the mLRS lua script work for you?

did you select the dip swithces correctly?

carefully go through the docs again

arg7, 2024-01-20:
Thanks for your support, I have done OpenEdge upgrade, without success, and checked and rechecked wiring, switches and so on.
And finally I have found the cause:
I wanted to test first 19hz mode and set "Rx Ser Baudrate" to 38400. Also in Mission Planner I have set same baud rate.
But I didn't noticed that it reverts to it default value 57600. This looks like a bug.

Since the baud rates on flight controller and RX didn't match, I was getting no telemetry.

Once I set 57600 on FC, it began works properly.

By the way, your project is amazing. Looking to contribute to it in the near future.

Best regards,
Andrey

olliw42, 2024-01-20:
> I wanted to test first 19hz mode and set "Rx Ser Baudrate" to 38400. Also in Mission Planner I have set same baud rate.
But I didn't noticed that it reverts to it default value 57600. This looks like a bug.

thx for reporting. Could you give me a recipe which I can follow to reproduce the bug? 
You may imagine that I do change parameter settings kind of every day, and it works, so it seems to happen only under certain conditions, which I may not easily catch. 

Does this reset happen only upon firmware flashing/upgrading, or are you saying you can't change the settings with the CLI or the Lua script?

arg7, 2024-01-20:
> Did you maybe forgot to save/store? In the cli it would be the command store;

Yes, you are exactly right. I didn't notice "Save" button in main screen, my fault.

Best regards,
Andrey

olliw42, 2024-01-20:
don't worry, no problem. 
It tells though we shoudl improve the docs a bit to mention this
So, I read this to mean that it's working fine now for you, great.