olliw42/mLRS #184: LUA script binding behaviour

joosttromp, 2024-05-21:
Steps to reproduce:
Run the mLRS LUA script (in my case on a TX16S color screen) and initiate binding. With the "Binding..." pop-up in view, press the back button.

Current behaviour:
When back button is pressed the lua script exits, but the mLRS module is still looking to bind. When the LUA script is re-started, the default view is displayed and the "Binding..." popup is not in view. When a receiver is powered up and set in bind mode, the 2 modules will still bind.

Expected behaviour:
When binding is initiated (but does not complete for some reason,) and the user presses the back button, the LUA script/module should stop trying to bind (and pop-up disappear). Then, only when the back button is pressed another time, the LUA script should exit.

olliw42, 2024-05-22:
many thx for the report. Much appreciated. 👍 

I looked a bit at it. There is no way currently to cancel a bind operation, besides binding to a receiver or by repowering. I actually think this is acceptable behavior, and that the additional code & complexity for a cancel mechanism isn't so much warranted.

That is, the "expected behavior" isn't easily achieved. The lua script behavior is however not ok, as you correctly state. I guess my proposal is that when the lua script is started it should jump into the binding page if the tx module is in binding mode.

joosttromp, 2024-05-22:
no problem at all!

Yes, that would then be the most elegant solution i think.

olliw42, 2024-05-22:
not so straightforward either since we also do not have code to tell the lua that the tx module is in binding mode LOL ... but it sounds much easier to do

as you als appear tlo find that solution acceptable, I guess that's what we are going to do then.

olliw42, 2024-10-21:
with latest firmware, lua script jumps to binding when tx module is in binding at start of lua.
So, resolved, hence closing.

closed 2024-10-21 by olliw42