olliw42/mLRS #190: TX firmware for my R9M Lite Pro?

Rothammel, 2024-06-01:
I can't find the TX firmware for my R9M Lite Pro.
in the firmware folder there is only tx-R9M-f103c8-v1.2.04.hex
Can I use this?

Thank you for your help

olliw42, 2024-06-01:
there is no such firmnware as mLRS does not support the R9M Lite Pro as tx module.
See https://github.com/olliw42/mLRS-docu/blob/master/docs/FRSKY_R9.md#r9m-lite-pro-as-receiver

If you want to use it as tx, you would have to compile your own firmware.

You cannot use tx-R9M-f103c8-v1.2.04.hex.

Sorry, sir.

olliw42, 2024-06-01:
closing. If you have further question on this, pl do not hesitate to reopen. :)

closed 2024-06-01 by olliw42

Rothammel, 2024-06-01:
it says in the readme:

"The code currently supports:

Frsky R9M and R9M Lite Pro transmitter modules and R9 MX, R9 MM and R9 Mini receivers (868/915 MHz)"

oh no, I only have a frsky taranis x9 lite.

then I would have to buy a new remote control, what would you recommend?
a radiomaster tx16?
then I could use the r9m lite pro as a receiver.

or should I refrain from using mLRS and install additional 900mhz telemetry?
I have 10 inch copters and large fixed wing models

olliw42, 2024-06-01:
you are correct, the wording is missleading. Raised an issue so we do not forget to correct. Thx for reporting. :)

I think I'm not in a position to make any recommendation. 

I'm not familiar with that radio, you may consider going to the discord channel, the folks there will have better ideas than me on what one may possibly do. I guess there could be a sort of a full Jr bay to micro bay converter box, or so, for e.g. using a R9M.

if you want to try mLRS or feel you better go with another 900 mhz system is really not for me to comment on. I'm obviously biased :)