olliw42/mLRS #200: bug: esp32 diversity, allow antenna2

olliw42, 2024-06-20:
on the esp32 receivers with diversity, which use single spi, selecting antenna2 will make it crash. This is attributed to how the spi and sx are initialized.
Needs to be corrected however.

brad112358, 2025-01-30:
Is there any chance this is related to https://github.com/olliw42/mLRS/issues/245 ?

olliw42, 2025-01-30:
this one here - if I'm not mistaken - is understood. If I recall it is because with only sx2 the spi is not initialized. We would need an underlying level of structure which does the spi init either case, but not twice. We don't have that, and a straigth fiorward apporach would look ugly, in terms of arhcitecture.

jlpoltrack, 2025-01-30:
Yes, unrelated.  Further - antenna 2 is disabled for all of the ESP32 diversity hardware. https://github.com/olliw42/mLRS/pull/203