olliw42/mLRS #201: No build E77 TX *(

Dwakener, 2024-06-25:
In file included from C:/mLRS/mLRS/CommonTx/mlrs-tx.cpp:50:
C:/mLRS/mLRS/CommonTx/../modules/stm32ll-lib/src/stdstm32-uartb.h:459:4: error: #error No UARTB TX IO defined!
  459 |   #error No UARTB TX IO defined!
      |    ^~~~~
C:/mLRS/mLRS/CommonTx/../modules/stm32ll-lib/src/stdstm32-uartb.h:462:4: error: #error No UARTB RX IO defined!
  462 |   #error No UARTB RX IO defined!
      |    ^~~~~
In file included from C:/mLRS/mLRS/CommonTx/mlrs-tx.cpp:71:
C:/mLRS/mLRS/CommonTx/../Common/mavlink/fmav.h:19:10: fatal error: out/mlrs_all/mlrs_all.h: No such file or directory
   19 | #include "out/mlrs_all/mlrs_all.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [CommonTx/subdir.mk:29: CommonTx/mlrs-tx.o] Error 1
make: *** Waiting for unfinished jobs....
"make -j4 all" terminated with exit code 2. Build might be incomplete.
[image]

olliw42, 2024-06-25:
@Dwakener 
just tested on my pc and this target compiles fine
there is most likely either something not complete and correct with your setup or you made changes which aren't correct. With the given info we cannot tell you.
I just note that in your main.cpp I do see a `return main()`, which isn't the original code. So it seems you made changes.

btw, this github issue area is NOT for support. The issue reported here cannot be confirmed and reproduced, so I'm going to close it, If you feel you desire further support with your issue, pl consider choosing the suppoprt channels, either at rcgroups or at discord (discord should be preferred since mor epeople are there)

thx for reporting

closed 2024-06-25 by olliw42

Dwakener, 2024-06-25:
Are you on Linux?

olliw42, 2024-06-25:
no, but it doesn't matter
other folks use it on linux

and your change to the code is just wrong

brad112358, 2024-06-25:
The master branch of tx-E77-MBLKit-wle5cc builds without errors or warnings on Linux.

olliw42, 2024-06-25:
MANY thx.