olliw42/mLRS #205: List of PA/LNA for targets

olliw42, 2024-07-03:
not an issue
I'd like to compile a list of the PA/LNAs used by the various targets. It's first to keep track, but also becasue I would like to better separate this out in the hal files.

- E28: SKY85321-11
- FRM303: SKY66312-11
- FrSky R9M: SKY65111-384LF
- FrSky R9M Lite Pro: SKY65111-384LF
- ELRS Generic 2400 D PA ESP8285: ??
- ELRS Generic 2400 PA ESP8285: ??
- ELRS Generic 900 PA ESP8266: ??
- ELRS Generic 900 TD PA ESP32: ??
- RadioMaster RP4TD: ??
- RadioMaster Bandit Micro: SE2435L

jlpoltrack, 2024-07-03:
> * ELRS Generic 2400 D PA ESP8285: ??

SE2431L

> * ELRS Generic 2400 PA ESP8285: ??

AT2401C

> * RadioMaster RP4TD: ??

AT2401C

olliw42, 2024-07-03:
THX!

you could double check/confrim some of the others I did already?

olliw42, 2024-07-03:
I'm a bit wondering on which occasion I stumbled over the SE2431L before ... its datasheet is for >2 years in the doc folder ...

jlpoltrack, 2024-07-03:
> you could double check/confrim some of the others I did already?

The others look good to me.  Suppose I could note that there are two E28 variants (mLRS designs use the 27 variant):

1. E28-2G4M27S/SX: SKY85321-11
2. E28-2G4M22S: AT2401C

jlpoltrack, 2024-07-03:
> * ELRS Generic 900 TD PA ESP32

Skyworks 66421-11

olliw42, 2024-07-19:
@jlpoltrack I've tried to compile things into a hal-power-pa.h hal file for later use, but it's pretty incomplete, and uncovers some inconsistencies. Thoughts? Corrections?

[hal-power-pa.zip](https://github.com/user-attachments/files/16303750/hal-power-pa.zip)

closed 2024-09-03 by olliw42