olliw42/mLRS #207: Lua TX12

Dwakener, 2024-07-10:
Good afternoon. TX12 hardware gets stuck loading LUA.  TX = E77-MBLKit-wle5cc

Dwakener, 2024-07-10:
10:15:51 : Data read successfully
10:15:51 : Time elapsed during the read operation is: 00:00:02.783
10:16:10 : Read File: C:\mlrs\firmware\pre-release-stm32\tx-E77-MBLKit-wle5cc-400-v1.2.04.hex
10:16:10 : Number of segments: 1
10:16:10 : segment[0]: address= 0x8000000, size= 0x13268
10:16:17 : Memory Programming ...
10:16:17 : Opening and parsing file: tx-E77-MBLKit-wle5cc-400-v1.2.04.hex
10:16:17 :   File          : tx-E77-MBLKit-wle5cc-400-v1.2.04.hex
10:16:17 :   Size          : 76.60 KB 
10:16:17 :   Address       : 0x08000000 
10:16:17 : Erasing memory corresponding to segment 0:
10:16:17 : Erasing internal memory sectors [0 38]
10:16:18 : Download in Progress:
10:16:19 : File download complete
10:16:19 : Time elapsed during download operation: 00:00:02.208

Dwakener, 2024-07-10:
[image]
[image]

olliw42, 2024-07-10:
many thx for the report
the lua support on bw radios is always a bit of an issue, since no dev has such a radio. And there are so many of them.

yo are running mLRS-bw, and not mLRS, right?

what firmware is your radio on? You need latest OpenTx or latest EdgeTx. 
do you have any other script or similar running, which would eat memory away?

Dwakener, 2024-07-10:
Expressel modules work stably. Regular script mLRS restarts the equipment,  mLRS-bw the loading text hangs and nothing happens. Out of curiosity I watched the broadcast, there is silence on 433MHZ

Dwakener, 2024-07-10:
I also looked at the jumper on the antenna, you didn’t have it in the pictures, here it is
[image]

Dwakener, 2024-07-10:
EDGETX-TX12MK2 2.8.0 CFGV 221

olliw42, 2024-07-10:
upgrade your edgetx to latest, it's known t not work on that old version

Dwakener, 2024-07-10:
edgetx-firmware-v2.10.2 helped !!!

closed 2024-07-10 by Dwakener

olliw42, 2024-07-10:
great
glad to hear it works

two general comments pl
- this was obviously a support topic and not an issue report. The github repo is NOT for support. For support pl consider going to the support channels next time (rcgroups thread or discord channel)
- consider readin and rereading the docs carefully in case of issues. One would have found the comment on the edgetx version there :) I mean, you sure are most welcome to raise your questions and ask for help, but it may safe you your time :)