olliw42/mLRS #218: Receiver not connected after signal lost

chphoto80, 2024-10-16:
Yesterday, I fly my plane with Ardupilot and mLRS system. I use Radiomaster TX16S + Matek mLRS 2G4 module and iFlight 2400 TD receiver. Firmware 1.3.03 in module and receiver. Operation mode 50Hz. I fly 11km and I lost signal of RC (I thing, not problem, Ardupilot takes plane back). When was plane above Home, I have no RC connection to plane, TX module blinks red. I must wait to discharge battery in plane and plane done hard landing. Not fatal. When I open plane, receiver blinks red. I disconnected powewr, re-connect and receiver still blinks red.

When I went home, I take receiver out of plane and done some tests "on table". I was able to simulate same issue as on field. I try to switch on receiver iFlight, receiver after seconds connected to TX module. I set minimal RF power and go with Radiomaster far away. Radiomaster says Telemetry lost. I wait few minutes and go back. Still no Telemetry and when I go home, receiver blinks rapidly red. No connection. Same situation as on field. I decide to try Matek RX and done same test. No issue, connection after signal lost done.

After it, I switch both receivers to operational mode 19Hz and done same tests. In this mode, no issue.

I think that according to the tests carried out, the problem is in the combination of the iFlight receiver and the used operation mode.

Before mLRS, I fly same receiver with ELRS, fly 15km and no problem. Therefore, I think that the problem is not in the HW (not faulty receiver), but in the mLRS software.

tmcadam, 2024-10-16:
@olliw42 There was some chat back around 20th June in General channel on discord around the iFlight receivers. I can't work out what the outcome was though. I think it was around accuracy of power levels rather disconnect issues like this.

olliw42, 2024-10-16:
MANY THX for the report. This is really much appreciated.

Am I right in assuming that you are the friend of discord user Osvetitel was mentionining? https://discord.com/channels/1005096100572700794/1005096101239603232/1295909073031794778

We are very sorry for the bad experience you had/have. And will be really eager to understand it better.

Thanks also for the good details of info. The "good" thing is that you seem to be able to reproduce it, this really gives us a handle.

v1.3.03: 
It's the v1.3.03 from the github prerelease repo, [rx-iflight-elrs-td-2400-v1.3.03-@891ce0b5.bin](https://github.com/olliw42/mLRS/blob/main/firmware/pre-release-esp/rx-iflight-elrs-td-2400-v1.3.03-%40891ce0b5.bin) on the receiver, and [tx-matek-mr24-30-g431kb-default-v1.3.03-@b302689c.hex](https://github.com/olliw42/mLRS/blob/main/firmware/pre-release-stm32/tx-matek-mr24-30-g431kb-default-v1.3.03-%40b302689c.hex) on the tx module?
Or is it self-complied?

receiver needs to be powered up:
that's a good piece of info and helps us a lot

matek receiver no issue at home:
that's also a good piece of info and helps us a lot

red LEDs:
also very good piece of info. The fact the leds blink tell that this was at least not a crash.
you say after the issue both the tx module and receiver show a red led. 
tx module: 
in both cases (flight and at home) the red led was doing the same? It did a 2 Hz blink as when disconnected?
receiver:
In the at-home test you say the receiver blinks rapidly red. 
Do you happen to recall if the receiver's RED led was blinking slow (ca 2 Hz as when just disconnected) or also rapidly (ca 5 Hz or so) in your first instance of the issue?
Was the green led on, or just the red led did something?

Some background: In the dev versions we do have a "FAIL" mechanism enabled to catch issues we think should not happen. This make flying somewhat less safe since when such a fail happens the device indeed goes into a blocking loop. The places where a red blink would be issues are these
https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L677-L679
https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L713-L715
https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L549

> I disconnected powewr, re-connect and receiver still blinks red.

if the receiver's RED led was blinking at 5 Hz this would indicate an issue with detecting the sx chips ... https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L549

so far for the moment
I think I do have an iflight TD somewhere, maybe I'll have a chance to try to reproduce on the weekend

olliw42, 2024-10-16:
> @olliw42 There was some chat back around 20th June in General channel on discord around the iFlight receivers. I can't work out what the outcome was though. I think it was around accuracy of power levels rather disconnect issues like this.

thanks! 
hmhm ... I believe to recall that we were for a short moment wondering if we shouldn't better take them off support ... let's see

chphoto80, 2024-10-16:
Hallo,

I will try to answer all questions below.

1) Yes, Osvetitel is my friend and this is my issue with iFlight receiver.

2) Matek TX have firmware from github and it is this file : tx-matek-mr24-30
***@***.*** Firmware was flashed by STM program 
(as is on Matek site); iFlight have firmware by github downloaded to my PC 
as is manual on git and firmware is ***@***.***
5.bin. To receiver it was fleshed by manual on git.

3) At home, Matek receiver has another issue (no GPS M10Q fix), but no 
problems with reconecting after signal lost

4) LEDs : When I fly and have problem, TX module blink red about 2Hz (as 
clasic when waiting for receiver). What done LED in plane, I dont know (in 
fly). After landing, LED was red, but I dont know, if flashing 2Hz or 5Hz. I
think, that was 5Hz also as home. In home, TX flashing 2Hz, but receiver 
flashing quickly, may be 5Hz. 

5) LED after re-powering : On field, I try to check, if conection will be 
established or not. LED on receiver was red, but I dont know, how blinking. 
But Radiomaster not show telemetry. After seconds I unpluged battery. At 
home, I put reveiver out of plane and tested. I expected only RED, but after
few seconds gone GREEN and Radimaster says low RSSI, disconecting telemetry,
telemetry ok and everything OK.

Best regards Tomas.

---------- Původní e-mail ----------

Komu: olliw42/mLRS ***@***.***>
Kopie: chphoto80 ***@***.***>, Author ***@***.***>
Datum: 16. 10. 2024 8:45:48
#218)
"

1005096101239603232/1295909073031794778
(https://discord.com/channels/1005096100572700794/1005096101239603232/1295909073031794778)

(https://github.com/olliw42/mLRS/blob/main/firmware/pre-release-esp/rx-iflight-elrs-td-2400-v1.3.03-%40891ce0b5.bin)
on the receiver, and ***@***.***
(https://github.com/olliw42/mLRS/blob/main/firmware/pre-release-stm32/tx-matek-mr24-30-g431kb-default-v1.3.03-%40b302689c.hex)

https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L677-L
679
(https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L677-L679)
https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L713-L
715
(https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L713-L715)
https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L549
(https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L549)
" 
mLRS/CommonRx/mlrs-rx.cpp#L549
(https://github.com/olliw42/mLRS/blob/main/mLRS/CommonRx/mlrs-rx.cpp#L549)

(https://github.com/olliw42/mLRS/issues/218#issuecomment-2415875604), or 
unsubscribe
"

jlpoltrack, 2024-10-16:
@chphoto80  Thanks for the details, few more from me:

- Could you share what power level on the Rx you were using?
- Can you also reproduce at min power?

Am wondering if heat plays a role here.

chphoto80, 2024-10-16:
Critical flight : TX power 1.0W, RX (iFlight) power 250mW - RX fail

Table test : TX power min., RX (iFlight) power 10mW - RX fail

Table test : TX power min., RX (Matek) power min. - RX not fail - OK

About heating, I dont know, but on min power at table was RX only little 
warm, not hot.

In my case, I think, heat is not cause of the problem.

---------- Původní e-mail ----------

Komu: olliw42/mLRS ***@***.***>
Kopie: chphoto80 ***@***.***>, Mention ***@***.***>
Datum: 16. 10. 2024 13:48:03
#218)
"

   * Can you also reproduce at min power on the Rx?

olliw42, 2024-10-16:
@chphoto80 

MANY THX for the additional info. Very helpfull.

> 3) At home, Matek receiver has another issue (no GPS M10Q fix), but no
problems with reconecting after signal lost

let me first comment on this. This is most likely due to the high power of the receiver. The GPS and receiver antenna need to be well separated. Sadly today often these very little GPS modules are used which nearly do not have any ground plane.
A good approach is to place the GPS at some high position, and the antenna rather at the bottom of the vehicle.

As regards the main issue:

I found my iFlight 2.4 TD. So I flashed it with the lartest firmware of today, and I also installed my Matek 2.4 Tx kit, also freshly flashed with todays firmware, into my TX16S. So, firmwares are not exactly yours but I can't see a releveant change compared to yours. 
I've set the power on the iFlight to 250 mW, and the power on the Tx module to min (lua script says 13 dBm).
The iFlight gets quite warm, touchable but pretty warm (it's also pretty warm when not transmitting, so it really has a poor heat design).
Anyway, went out with the transmitter, and walked until I got connection loss, went back until I had connection, and I did this 10 times .... each time it would reconnect, then went back home, and receiver was still working fine.

So, sadly, I cannot reproduce your issue.

There is one thing I'd like you to test: On the receiver you can change the "Diversity" field from "enabled" to "antenna1" or "antenna2", can you test for each "antenna1" and "antenna2"?

Besides that, I have to report that my iFlight also behaves sometimes dodgy ... I could vbting it into a condition where after power up it would just continuously go through restarts (three fast red leds, pasue, 2x fast leds, etc). So I tested also with my RPTD4, and this doesn't give me such issues.

chphoto80, 2024-10-17:
Hallo,

about iFlight receiver, what Operational mode was set? I have 50Hz and when 
I changed to 19Hz, receiver connected. 

I could vbting it into a condition where after power up it would just 
continuously go through restarts (three fast red leds, pasue, 2x fast leds, 
etc).- Yes, this happened 2 times me too.

And now about GPS.

On my plane, I have GPS at the end of plane (AtomRC Killer Whale) beetween 
rudders. iFlight reveiver and his antenas is about 10cm far. With iFlight 
receiver NO problem with GPS. But with Matek receiver, no GPS fix when 
receiver is at the same place as iFlight. I do many test and when I 
disconected power from Matek RX, GPS will fix. After fix, when I reconnect 
power to Matek, GPS is fix. But when is plane home, I go out and power up 
plane, I wait 12 minutes and no fix. Only, when I put receiver very far from
GPS (on long cables), GPS fix. But this is not solution, plane is not big to
do it as solution for fly. This was a motive to use iFlight reveiver and not
Matek for test flight - GPS.

---------- Původní e-mail ----------

Datum: 17. 10. 2024 1:48:18
#218)
"

   1. At home, Matek receiver has another issue (no GPS M10Q fix), but no

olliw42, 2024-10-17:
oh, yes: 50 Hz

chphoto80, 2024-10-17:
Today afternoon, I will try to make video with my receiver and all my 
setting.

By the way, I used rc override, did you too?

---------- Původní e-mail ----------

Datum: 17. 10. 2024 7:50:49
#218)
"

olliw42, 2024-10-17:
quick question concerning GPS: 
what power did you have on teh matek receiver for the issue? Also exists with min? 

> I used rc override, did you too?

not for the tests I described, shouldn't matter

chphoto80, 2024-10-17:
I tested GPS with 1.0W TX and receiver 250mW, minimum I dont tested. May be 
in the afternoon test too with RX fail.

---------- Původní e-mail ----------

Datum: 17. 10. 2024 8:15:09
#218)
"

olliw42, 2024-10-17:
a test was just made with a GPS like 3 cm next to a 2.4 GHz dipole antenna and the m24-30 receiver like 5 cm away from the GPS and reciver at 1W, tx at 100 mW, and no issue
so it doesn't appear to be like a general problem
looking forward to your further info

chphoto80, 2024-10-17:
And what system of test you done. Because :

When I do cold start GPS (plane is at home for 1 day), no fix. But when I do
test after fix (at same day, e.g. hot start) there is no problem with fix.

Problem is only first fix with cold start.

--------- Původní e-mail ----------

Datum: 17. 10. 2024 8:37:03
#218)
"

MATEKSYS, 2024-10-17:
Just did a test,  put mR24-30(1W RF) & M10Q-5883 GPS neayby window.  after powering on both at same time,  bridge M10Q Reset to G to trigger cold start.  GPS can get 3D fixed after a few minutes.
[image]

chphoto80, 2024-10-17:
Now, I done test and this is video of my test.

As is see in video, Matek mLRS receiver is about 10cm far from Matek GPS. 4 
minutes and no fix. After disconecting power from receiver, GPS fix after 1 
minute. (green led on GPS flash).

https://www.youtube.com/watch?v=fUPxyCrxwd4

And here is video of setting in Radiomaster

https://youtube.com/shorts/nuSZhNl5jDE

Best regards Tomas.

---------- Původní e-mail ----------

Datum: 17. 10. 2024 12:25:03
#218)
"

image.png (view on web)
(https://github.com/user-attachments/assets/f976f39e-48fa-4b7a-ae28-8cbe87a2c00e)

olliw42, 2024-10-18:
> Now, I done test and this is video of my test. As is see in video, Matek mLRS receiver is about 10cm far from Matek GPS. 4 minutes and no fix. After disconecting power from receiver, GPS fix after 1 [...]

many thx for the video
I fully agree that this looks strange, and I can't offer an explanation.
I just know that on the other hand it's not for all like this. 
@MATEKSYS any ideas?

I guess if I were you I would assume for the moment that it is maybe not the receiver but something else.
The wiring to the GPS was mentioned. I guess what I would try is to keep everything as is, but unplug the GPS and power it from a small battery close to it. Just to see if that has an effect.

MATEKSYS, 2024-10-18:
> many thx for the video I fully agree that this looks strange, and I can't offer an explanation. I just know that on the other hand it's not for all like this. @MATEKSYS any ideas?
No idea.  Maybe wait a few more minutes and see what happens.

chphoto80, 2024-10-18:
Re : No idea. Maybe wait a few more minutes and see what happens.

And what is FEW minutes, is for you 15 minutes OK. I test it on field, I 
wait for 15 minutes and no fix. Disconect receiver and fix after 1 minute.

---------- Původní e-mail ----------

Datum: 18. 10. 2024 8:02:23
#218)
"

" "

chphoto80, 2024-10-18:
Re : I guess if I were you I would assume for the moment that it is maybe 
not the receiver but something else.......

I cannot agree with your opinion. This plane with this GPS module, with same
FC and same wiring, fly with ELRS (with same iFlight receiver 2400TD) about 
100km track (more flights, not only one). No problem with receiver, no 
problem with GPS, no problem anywhere. I only change ELRS to mLRS and many 
problems starts. For me, I try mLRS because TX module have UART (ELRS not 
have this). But now, I change the decision and return to ELRS.

---------- Původní e-mail ----------

Datum: 18. 10. 2024 7:50:18
#218)
"

"

geofrancis, 2024-10-18:
I think it's a hardware issue you're having as it can't be replicated, I would get a replacement.

chphoto80, 2024-10-18:
Problem may be in TX module, because RX have 2 pcs of Matek, one iFlight 
(critical flight) and yestarday I flashed Radiomaster RP4TD and this 
receiver Iam not able to pair with TX module. Iam very flustrated of this.

---------- Původní e-mail ----------

Datum: 18. 10. 2024 16:35:01
#218)
"

chphoto80, 2024-10-18:
You are right, TX, not RX.

---------- Původní e-mail ----------

Datum: 18. 10. 2024 19:47:11
#218)
"

I just did exactly this combo few days ago in the tests I described in the 
above
did you by chance flashed the RP4TD with the tx- firmware, not the rx- 
frmware? (did that mistake mysef initially)

chphoto80, 2024-10-18:
RP4TD flashed with right firmware, receiver connect to TX module on power 
on. iFlight receiver I must bind. Is it right?

---------- Původní e-mail ----------

olliw42, 2024-10-18:
the iflight, when flashed newly, should also not need binding

chphoto80, 2024-10-18:
And just on the first connection, iFlight was behaving strangely and 
connecting and disconnecting, so I had to put Bind.

I think, iFlight was bad choice for flight (with mLRS). May be TX module 
fails, RX fails, or something else.

---------- Původní e-mail ----------

Datum: 18. 10. 2024 20:09:03
#218)
"

the iflight, when flashed newly, should also not have needed binding

olliw42, 2024-10-18:
so with RP4TD it's all good?

chphoto80, 2024-10-19:
Now, I done all tests at home. I test it on my plane, with all equipment as 
I fly. RX antenas was about 10cm far from GPS (as Matek tests) and I have 
fix after 1 minute (in town as shown in video, same place) and this is 
normal for me. Power was 100mW on both. With GPS not problem with iFlight. 
Now with RP4TD not problem too.

After it, I switch pover to min at TX and 1mW at RX and went to far from 
plane. Signal lost after few meters and few walls, I wait for few minutes 
and go back. Signal recovery.

This receiver (RP4TD) did not have a power-on problem during tests like the 
iFlight receiver. So I think the iFlight receiver is not suitable for mLRS 
at all. RP4TD seems more useful for mLRS.

During my tests today, however, I noticed a 3x problem with the LUA script, 
which wrote a red warning and did not connect to the module and write try it
again. After reload script, everything was ok. This again reinforced my 
suspicion that my transmitter module was not quite right.

---------- Původní e-mail ----------

Datum: 19. 10. 2024 1:07:09
#218)
"

olliw42, 2024-10-19:
so, I read this to mean that it's working fine with the RP4TD

I still would kindly ask you, if possible, to do the battery test with the matek receiver I suggested in the above. Not to imply anything, but it really may help us to determine what's going on.

chphoto80, 2024-10-19:
Hallo,

I done 2 tests with GPS module and Matek receiver with battery. Receiver 
have one baterry, GPS second battery (as photo) and after powering and 1 
minute, GPS blinks as fix.

Second test was with only 1 batterie, but on power cables was ferrite filter
rings (as second photo). After powering and waiting 2 minutes GPS fix!!!

I think, this is solution.

---------- Původní e-mail ----------

Datum: 19. 10. 2024 11:37:07
#218)
"

olliw42, 2024-10-19:
thx a lot for these test

so, it appears to be some noise on the power line generated by the matek receiver

it could make some sense. As opposed to the other receivers in talk the matek has a SR for voltage regulations ... and at least in my junger days I've learned it's not good to power a SR form a SR (and the voltage regulators on FCs are commonly SRs). Just a wild theory, for sure, but at least some theory LOL

I think what you can do in addition to a ferrite ring is to twist the power wires, especially those to the GPS

maybe your flight controller also has a 5V and a 4.5V pin, and you can power the matek receiver from the 5V pin, may give another layer of isolation

chphoto80, 2024-10-20:
Thank you for information, It seems to be right. Ferrite ring is OK, cable 
twist too and I try it.

Now one question. Is better use rc override (only one cennection UART, 
Ardupilot number2), or use separate UART for RC (Ardupilot number 23 and for
Mavlink 2 on other UART)? When I use nonMatek receiver, there is no other 
choice - only rc override is possible. But with Matek receiver, I can use 
both connection. 

---------- Původní e-mail ----------

Datum: 19. 10. 2024 17:51:52
#218)
"

olliw42, 2024-10-21:
rc_override, this is a general support question and not so well placed here.

> Thank you for information, It seems to be right. Ferrite ring is OK, cable
twist too and I try it.

you're still in testing or can we conclude all is ok now, and that the issue could be closed?

chphoto80, 2024-10-21:
OK, may be closed.

---------- Původní e-mail ----------

Datum: 21. 10. 2024 7:39:28
#218)
"

olliw42, 2024-10-21:
glad to see it could be resolved
MANY THX

closed 2024-10-21 by olliw42