olliw42/mLRS #239: Matek mR900-30: can't flash firmware in STMCubePorgrammer

AndiFlyy, 2024-12-03:
Dear mLRS developpers,

I just bought 2 Matek mR900-30 and tried to flash the following firmwares:
- tx-matek-mr900-30-g431kb-default-v1.3.03-@b302689c.hex
- tx-matek-mr900-30-g431kb-siktelem-v1.3.03-@b302689c.hex
- rx-matek-mr900-30-g431kb-v1.3.03-@b302689c.hex

I followed the steps from MATEK homepage:
https://www.mateksys.com/?portfolio=mr900-30#tab-id-4
especially concerning DFU drivers:
https://www.mateksys.com/?p=5159#tab-id-1

Tried ZADIG and ImpluseRC Driver Fixer as described there, as well as STM32 Virtual Com Port driver.

I always end up with the same error in STMCubeProgrammer:
  "Address: 0x00000000  Error: Start operation failed"

It seems, that the erasing in STMCubeProgrammer worked, because the modules don't blink anymore, as they did when they arrived.

Maybe the problem is in the STM32 VCP Drivers. In the descrition it says, in windows device manager it must be displayed as COM port. This is not the case at my PC unfortunately. I see the mR900-30 as: 
>USB-devices > DFU in FS mode

Tried it on 2 different PCs already.

Any advice howto flash successfully?

Thank you! 
Best regards,
Andi

jlpoltrack, 2024-12-03:
> Any advice howto flash successfully?

Try this:
- Hold down the button while plugging in the USB cable, this will put the MCU in DFU mode
- Open Chrome, go here https://mlrs.xyz/flash/ and try to flash the firmware for your device

closed 2024-12-03 by AndiFlyy

olliw42, 2024-12-04:
@AndiFlyy please note that the github issues are NOT for support questions, that's what the threads at rcgroups or the discord channel are for. Please consider joining one of them (discord is more active) for further support questions. Thx for your understanding.

> On STMCubeProgrammer flashing, any crew why it didn't work? Anybody the same problems?
As it worked with the Web-Flasher now, that's the proof that the USB / VCP drivers were not the problem.

usually it's never a good idea to juggle with the drivers, they are usually all installed correctly when installing STMCubeProgrammer.

AndiFlyy, 2024-12-06:
Dear Olli,

get your point.

Some positive feedback for you anyway:
Did some first rangetest, 10km no problem at all!! Rocksolid connection, even with the ground antenna close to the ground.
Therefore my guess is that it goes way further!

Amazing work you did!!

Thank you so much!

Cheers,
Andi