olliw42/mLRS #245: Crash on DIY E28 dual board when diversity is disabled.

brad112358, 2024-12-13:
When I use mLRS-bw.lua to switch from diversity to antenna 1 or antenna 2 and power cycle, it seems to be crashing very quickly after power-up, leaving the red LED on solid.  Changing other settings and power cycling seems fine.  The SWUART debug port works fine  in diversity mode and I see a "." printed every second, but when I power cycle after selecting antenna 1 or antenna 2, I'm not getting anything at all on the debug port, just the solid red LED.  Note also that if I disconnect JRPIN5 it powers up OK and the red LED flashes as it should.

This issue does not appear when DEVICE_HAS_DIVERSITY is not defined.

brad112358, 2024-12-13:
It works in branch v1.3

brad112358, 2024-12-13:
I ran a git bisect.  It took about 8 steps.
~/git/mlrs$ git bisect bad
1ebfbc6fd384a7ae071ef95d8e4959ff22b04bf7 is the first bad commit
commit 1ebfbc6fd384a7ae071ef95d8e4959ff22b04bf7
Author: olliw42 <waldmanns@gmx.de>
Date:   Sun Sep 22 15:09:12 2024 +0200

    tx,rx, make rf power changeable without save/store

 lua/mLRS.lua                           |  9 ++++-
 mLRS/Common/rf_power.h                 | 61 ++++++++++++++++++++++++++++++++++
 mLRS/Common/sx-drivers/sx126x_driver.h |  6 ++++
 mLRS/Common/sx-drivers/sx127x_driver.h |  6 ++++
 mLRS/Common/sx-drivers/sx128x_driver.h |  6 ++++
 mLRS/Common/sx-drivers/sx12xx_driver.h |  1 +
 mLRS/CommonRx/mlrs-rx.cpp              |  4 +++
 mLRS/CommonTx/mlrs-tx.cpp              |  6 +++-
 8 files changed, 97 insertions(+), 2 deletions(-)
 create mode 100644 mLRS/Common/rf_power.h

jlpoltrack, 2024-12-13:
On E28 Dual, if I set to Antenna1 to end up in a crash state, then comment out the sx2.UpdateRfPower line here and flash things seem to work: https://github.com/olliw42/mLRS/commit/1ebfbc6fd384a7ae071ef95d8e4959ff22b04bf7#diff-d442ec9fc39334fee02e97533bc34e173409b31ffbd4f4e056bfd97b76cbb18cR50-R51

Some extra logic needed here?

olliw42, 2024-12-13:
good catch
one probably needs to check the antenna setting and only call which had been inited

olliw42, 2024-12-14:
@brad112358 @jlpoltrack 

it appears that's again one of the strange things. Tried to reproduce on my dualE28 diy receiver, and it didnÄ't gave issues...

anyway, here is a branch with a suggested change: https://github.com/olliw42/mLRS/tree/owpr-rfpower-antenna

pl check it out if it removes the issue for you

jlpoltrack, 2024-12-14:
Ah, perhaps one difference - Brad and I were testing the E28 Dual board as a Tx, not a receiver.

My test with that fork:

- Switch to Antenna1 in Lua, save, then power cycle - all good.  This would previously crash, much better.
- Switch to Antenna 2, save, then power cycle - looks good.  However, when I go into the Lua the Tx power reports as 9 dBm (min) and the Sensitivity as -104 dBm (FLRC) when I'm set to 50 Hz.  RF Power meter shows that the original power settings are still there, so this seems to just be a Lua/reporting issue?

olliw42, 2024-12-14:
> Switch to Antenna 2, save, then power cycle - looks good. However, when I go into the Lua the Tx power reports as 9 dBm (min) and the Sensitivity as -104 dBm (FLRC) when I'm set to 50 Hz. RF Power meter shows that the original power settings are still there for Antenna 2, so this seems to just be a Lua/reporting issue?

shouldn't this be a bug which should have existed before, and also should exist then for the rx side?

maybe the latter, i.e. what happens with a rx, could be tested too

brad112358, 2024-12-14:
I only have one of these and I was only using it on the Tx side.  In addition, I only saw the crash when the Tx was actually installed in a radio handset.  Powering up the Tx without Jr pin 5 connected did not crash.  Do we understand why it matters if Jr pin 5 is connected?

olliw42, 2024-12-17:
> Do we understand why it matters if Jr pin 5 is connected?

that's inded a very good question. I guess we would want to. :)

jlpoltrack, 2024-12-17:
Would think it has something to do with channel data being read, does this trigger the update power function?

jlpoltrack, 2025-01-28:
Update:

- The proposed branch (https://github.com/olliw42/mLRS/tree/owpr-rfpower-antenna) addresses the issue with the Tx hanging at power on with Antenna1 or Antenna2 selected.  
- Another method that I tried was to use tdiversity.antenna to update the power (https://github.com/jlpoltrack/mLRS/tree/single-antenna-power) this also addresses the issue of the Tx hanging and I think saves a little time for diversity setups since only one sx chipset has their power updated.
- There is a separate issue that exists on the commit prior to the power update (https://github.com/olliw42/mLRS/commit/38262511e183ce74f3fa6c578788e6842ff0a575) which is that when Antenna2 is selected, the Lua reports incorrect configuration information.  Note in photo below the Lua shows 50 Hz with -104 dBm sensitivity.  This is incorrect, the FLRC sensitivity.
[image]

Suggest that we go with either of the two options to address the hang at power on with a single antenna selected and then create a separate issue for this configuration one.

olliw42, 2025-03-06:
@jlpoltrack 
I've pushed a change to account for the wrong reported RF sensitivity. It's based on your code change in https://github.com/olliw42/mLRS/pull/266, with a slight change to account for all diversity cases correctly.

olliw42, 2025-03-06:
@brad112358 @jlpoltrack I think I've determined and resolved the issue, pl see https://github.com/olliw42/mLRS/pull/302

would be great to see some independent testing :)

olliw42, 2025-03-07:
closing, seems to be resolved with https://github.com/olliw42/mLRS/pull/302, https://github.com/olliw42/mLRS/commit/87ad627b2f14b9cc9a4e8a15de713fa9febde8d0

closed 2025-03-07 by olliw42