olliw42/mLRS #271: GSpd and real spd seem to be quite different

olliw42, 2025-02-01:
a user at discord rasied this

https://discord.com/channels/1005096100572700794/1005096101239603232/1334908491495772264

The value reported in the CRSF telemetry element GPsd seems to be quite different to the value of GS/AS reported in MissionPlanner HUD.

jiri-jirus, 2025-04-25:
hello, any follow on this? do you need to provide some more evidence? thanks

olliw42, 2025-04-26:
@jiri-jirus 
no. It's also because the user who reported wasn't totally sure there was an issue.

so, you see that issue too? If so, can you provide info?

olliw42, 2025-04-26:
if it's of any help

the CRSF GSpd data comes from the mavlink message VFR_HUD.groundspeed field
https://github.com/olliw42/mLRS/blob/main/mLRS/CommonTx/crsf_interface_tx.h#L675-L703
https://mavlink.io/en/messages/common.html#VFR_HUD

so we need to understand where MP's AS/GS come from ...

olliw42, 2025-04-26:
arghh ... stupid me ... one of course needs to multiply with 3.6 and not devide to get from m/s to km/h ... grrr

just tested by faking an vfr hud speed, and somehow "km/h / 100" seems to mean in units of 0.1 and not in units of 0.01 ... ?!?!?! is this a bug in teh docs??

anyway, if we asume it needs to be  10.0 * vfr_hud_groundspd_mps * 3.6, then 
(1) I do get the correct GSpd shopwn in my telemetry
(2) the obeservation of the user makes sense since 14.1 * (100/3.6)/(10*3.6) = 10.88 !!

so, we shoudl conclude that conversion formula, I guess

olliw42, 2025-04-26:
ups ... AP also does 100 / 3.6 ... LOL
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RCTelemetry/AP_CRSF_Telem.cpp#L1017
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.h#L200
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.h#L367-L373

so mLRS at least is doing what AP is doing ...

olliw42, 2025-04-26:
given that the above new conversion formula gives the correct value in EdgeTx I call it a day, merge the change, and call this closed here :)

closed 2025-04-26 by olliw42

jiri-jirus, 2025-04-26:
thanks for fix. sad thing is that i am probably only user of mlrs who cares about telemetry

olliw42, 2025-04-26:
> thanks for fix. sad thing is that i am probably only user of mlrs who cares about telemetry

why would you think this