olliw42/mLRS #288: power switch synchronity issue !?

olliw42, 2025-02-10:
for our info

https://www.rcgroups.com/forums/showthread.php?4037943-mLRS-Lora-based-Mavlink-oriented-open-source-radio-link/page49&perpage=50#post53467985

[QUOTE=CG Photo;53467985]First flight today with mLRS. Flying with Matek 915mhz system. Did a maximum range test of 7.4km on 50mw at 19hz before it went to failsafe using standard dipoles on both ends. Best telemetry data rate I have ever seen on my tablet display running qGroundControl monitoring Autopilot.  The tablet never lagged at all. This was done in a high noise area. A DragonLink 915mhz system was only able to go 4km over this same area at full 1W power. 

Total new user here but I have a couple of thoughts. The first is a very minor bug in the way the power switch is monitored. If the power switch is the low power position and I start up the radio, the mLRS does not notice and sets the power to the higher setting. The power will go down after I toggle the switch. Also after some link lost events (like if I turn off the receiver and turn it back on) the power seems to get bumped up without the switch position being changed and then the radio just stays at the high power setting even with the switch in the low power setting. Seems that the power switch position needs to be looked at by the program more often. 

Second thought is a feature request. It would be handy to have a setting on the receiver power setting to follow the output power of the transmitter. So if I set the transmitter power to 250mw, then the receiver also goes to 250mw.  eLRS recently added this feature so you don't have to keep going in and changing the power the receiver for different types of flying.[/QUOTE]

Target0815, 2025-02-10:
To check: Start the mLRS LUA script, 30 dBm is always displayed for TX, even if the power switch is set to low.

Power up, power down, ReLoad - then the correct 17 dBm is displayed.

jlpoltrack, 2025-02-10:
@Target0815 Can you try this firmware for Matek mR900 Tx?  Should resolve the issue.

[tx-matek-mr900-30-g431kb.elf.zip](https://github.com/user-attachments/files/18734844/tx-matek-mr900-30-g431kb.elf.zip)

Target0815, 2025-02-10:
> [@Target0815](https://github.com/Target0815) Can you try this firmware for Matek mR900 Tx? Should resolve the issue.

This solves the problem.

olliw42, 2025-02-10:
thx to @jlpoltrack 
issue is supposed to be resolved now
so closing

closed 2025-02-10 by olliw42