olliw42/mLRS #297: Bug: Channel Data broken on EdgeTx 2.11

jlpoltrack, 2025-03-02:
mLRS can no longer understand channel data over CRSF on EdgeTx 2.11 (tried RC2) whereas 2.10.6 works.  The Lua works fine but no channel data is sent to the receiver and switchable power also doesn't work.

Guess is that somehow related to these PRs:

https://github.com/EdgeTX/edgetx/pull/5641
https://github.com/ExpressLRS/ExpressLRS/pull/3008

olliw42, 2025-03-02:
really? are they going nuts now?

can you in ad nutshell say what they do?

jlpoltrack, 2025-03-02:
There's now an optional byte in the RC Channels frame that ELRS uses for arming so you can get channel 5 back.  What I don't understand is that this should be backwards compatible and this extra byte should not be there unless you select it from the EdgeTx GUI.  The relevant code I think would be here: https://github.com/EdgeTX/edgetx/pull/5641/files#diff-b734ecd2cf7abf1d2157b23e4e59d166c2407b29afba5c4e493eb8e25324016f

On the scope, I checked the CRSF output and it was still 26 bytes so not 100% sure what is going on.

olliw42, 2025-03-02:
so, basically, they break the official CRSF spec here?

olliw42, 2025-03-02:
@rotorman Risto, any idea why this may break us?

olliw42, 2025-03-02:
a wild guess, maybe it's this thing https://github.com/EdgeTX/edgetx/blob/main/radio/src/telemetry/crossfire.cpp#L303-L306, i.e., that the fields are not properly reset/initialized if the tx module doesn't do device info.

jlpoltrack, 2025-03-02:
Okay, I have the scope hooked up with an mLRS module and it looks like the radio will just continually spam 0x28 'Parameter Ping Devices' https://github.com/tbs-fpv/tbs-crsf-spec/blob/main/crsf.md#0x28-parameter-ping-devices and not provide any channel data.

olliw42, 2025-03-02:
ah, so maybe it's not an RC thing at all but they have changed the behavior here ...

anyway, that's a ping to C8, i.e., the flight controller ... did edgetx not do it before?? I'm confused ...

jlpoltrack, 2025-03-02:
I think this PR: https://github.com/EdgeTX/edgetx/pull/5506

How I read the message:

- C8 is listed as 'Serial sync byte' in the CRSF spec
- 04 - Length
- 28 - Ping Devices
- 00 - Dest Address | Broadcast
- EA - Origin Address | 'Remote Control'
- 54 - CRC

jlpoltrack, 2025-03-02:
I think the logic on EdgeTx is to query for device information until a response before any channel data is sent: [[https://github.com/mha1/edgetx/blob/main/radio/src/pulses/crossfire.cpp#L180-L181](https://github.com/EdgeTX/edgetx/blob/main/radio/src/pulses/crossfire.cpp#L179-L180) so I guess mLRS will need to respond to the message...

olliw42, 2025-03-02:
this is then indeed a change in behavior on edgetx side 
do we have a PR where this was made?

jlpoltrack, 2025-03-02:
> do we have a PR where this was made?

Did you see this?  Posted few comments ago :) https://github.com/EdgeTX/edgetx/pull/5506

olliw42, 2025-03-02:
yes ... but I don't see that change there ... as much as I understand that PR it's about when to send the modeID crsf message ... i.e., I fail to see how it connects with what you just pointed at

jlpoltrack, 2025-03-02:
You're right, I took a closer look and this section was added in 2.11:

https://github.com/EdgeTX/edgetx/blob/2.11/radio/src/pulses/crossfire.cpp#L180-L184

Looking at 2.10 the pingFrame / bindFrame isn't there at all: https://github.com/EdgeTX/edgetx/blob/2.10/radio/src/pulses/crossfire.cpp#L130-L135

The pingFrame function looks to be the only function that can create these ping devices message...

olliw42, 2025-03-02:
this is the code of 2.10
https://github.com/EdgeTX/edgetx/blob/2.10/radio/src/pulses/crossfire.cpp#L130-L136
which is the same as in opentx
so, in 2.11 this piece has fundamentally changed ...

olliw42, 2025-03-02:
haha ... double work :)
so, I guess we know the issue now

jlpoltrack, 2025-03-02:
LOL - and for completeness, here is the PR where this was pingFrame / bindFrame was added: https://github.com/EdgeTX/edgetx/pull/4907

olliw42, 2025-03-04:
closing, PR seems to solve
THX!

closed 2025-03-04 by olliw42