olliw42/mLRS #308: MAVLinkX data loss/corruption of GPS_RTCM_DATA messages (?)

festlv, 2025-03-14:
I noticed weird behaviour of mLRS telemetry link when trying to send RTK GPS corrections from GCS to UAV side.
When the rx link type is set to mavlinx, I could observe some data passed to GPS module, but it refused to apply the corrections and stayed in 3d fix mode.
After setting the link mode to transparent, the GPS accepted the corrections (without changing anything else in the setup) and went into RTK fix modes.

The only tip for reproduction I have is: the link transfers several GPS_RTCM_DATA messages, approx 2KiB/s. Number of messages is around 10/s, but they are sent in quick succession.

I don't think I'll have enough time to do data dumps but I figured out that I should at least let others who are using mlrs for this know.

olliw42, 2025-03-15:
@festlv 
many thx for the report.

It is a bit weired though since the gps_rtcm_data message is in the lib and should be handled. What you are saying seems to say
1. these messages are transmitted but with some loss of packets
2. they are getting somehow corrupted so the gps can't digest anymore

Point 1 could be possible but I don't see how this should be different to then using transparent mode.
Point 2 I just don't understand.

Could you collect some more data?, 
- did you try mavlink instead of mavlinkX? Would be good to understand if this is a mavlinkX issue or an issue of your mavlink handling per see.
- do you have means to log the output on the mLRS receiver, e.g. with a usb-ttl adapter any pymavlink code, to see in which way the GPS_RTCM_DATA message could be corrupted?

Could you share your mLRS settings band, mode, mLRS hardware, mLRS firmware version, as well as your ArduPilot firmware version?

festlv, 2025-03-17:
1. I actually realized that by setting the transparent mode, I'm losing telemetry on the radio, so I did try mavlink link mode- it still does not work.
2. I have a test script that transmits the RTCM messages (as I was suspecting missionplanner as well)- I'll probably use that and log the messages that are transmitted and messages that are received.

Test setup is: 915MHz, 31Hz rate, Mateksys hardware, v1.3.03 fw, Ardupilot v4.5

olliw42, 2025-03-17:
thx for the aditional info
as said it's quite wired it would not work, so info of what arrives on the receiver side would be great
if the test sycript you mention is a py script you could share it, we might try it

festlv, 2025-03-18:
After a bit of further testing, it appears that there is increased data loss with mode = mavlink (vs transparent).
For this reproduction test, I am using a bit different hardware: 2.4GHz Mateksys, FLRC mode, no other devices connected (i.e. no RC transmitter), but the issue is still reproducible.

Here is a gist that contains:
a) self-contained python script (needs just pymavlink) that connects to two serial ports, transmits simulated GPS_RTCM_DATA traffic, and so far just counts the number of transmitted/received messages (RADIO_STATUS messages are ignored)
b) results for rx_ser_link_mode = transparent and mavlink 
c) mlrs configuration and versions.

To me it looks like the bursty nature of this kind of traffic is that stresses something too much- although the average data rate is ~1.5KiB/s, all of it is transmitted in short time (although, transparent mode does not mind).

olliw42, 2025-03-18:
MANY THX!

we need to investigate why it looses that many messages in the forward Tx->Rx direction

it should not matter that it's so bursty ... we do have a pretty big input buffer ... and especially it's hard to see why it shouldn't be a matter with transparent

olliw42, 2025-03-18:
EDIT: I think it would be useful to modify the py script to also log the seq number of the respective message ... this really tells much better what's missing :)

festlv, 2025-03-18:
> EDIT: I think it would be useful to modify the py script to also log the seq number of the respective message ... this really tells much better what's missing :)

Done, script + results updated in the original gist.

jlpoltrack, 2025-03-18:
@festlv Can you update to 1.3.04 firmware on both Tx and Rx and confirm you see the same?  Versions in gist show somewhat older firmware: 

[image]

Easiest way may be the Web Flasher: https://mlrs.xyz/flash/

jlpoltrack, 2025-03-18:
Curiously, the sequence number is out of order on both MAVLink and Transparent, just one example but there are many:

[image])

festlv, 2025-03-18:
> [@festlv](https://github.com/festlv) Can you update to 1.3.04 firmware on both Tx and Rx and confirm you see the same? 

I feel happy and sad at the same time- updating my desktop setup to 1.3.04 did indeed fix the issue. The issue was still present in v1.3.03 on 900MHz mlrs. I will now update the flying setup to 1.3.04 and report back.

jlpoltrack, 2025-03-18:
Hehehe, yes I missed the received / sending part, my bad.

olliw42, 2025-03-18:
great it seems to work with v1.4.04 ... I must admit I would not know why though ...

concerning the seq no: they seem not out of order
note this: sending is from the GCS, and it is spit out as fast as the GCS can spit it out. receiving is however on the rx side, and they will arrive as fast as mLRS can deliver them through the air. So, you in fact can see here the latency of transmission. The only factor of importance is that each seq no which you see on the GCS side shows up later on the rx side. In other words, the seq no on the rx side should be continuous without gaps.

jlpoltrack, 2025-03-18:
So just to confirm - the issue appears to be that mLRS only sends out the first 6 (seq 0 to 5) of the RTCM sequence?  6 through 10 aren't received?

﻿﻿﻿
[image]

olliw42, 2025-03-18:
I guess my first thing to look at would be to check the involved buffers and their sizes ... would be kind of the simplest explanation ... (although I thought we have big buffers ...)

(I'm on a trip for a couple of days and can't do anything than to talk LOL)

jlpoltrack, 2025-03-18:
> I guess my first thing to look at would be to check the involved buffers and their sizes

I would think it can't be the UART buffer since transparent mode works fine...

olliw42, 2025-03-18:
> > I guess my first thing to look at would be to check the involved buffers and their sizes
> 
> I would think it can't be the UART buffer since transparent mode works fine...

me too, but ... in the code I can't see either where it would get lost ...

jlpoltrack, 2025-03-18:
@festlv Few notes:
- Can you confirm you're using USB on the Tx?  (Based on the Py script I guess you are)
- Have you tried using the Serial 2 / LPUART pins?  If not, could you?
- More of an FYI for us, the USB RxBufSize is only 512 when I look here (am not sure if this gets overriddent): https://github.com/olliw42/mLRS/blob/main/mLRS/modules/stm32-usb-device/usbd_conf.h#L36-L37
- I did try increasing the USB RxBufSeize to 2048, but PyScript shows no difference
- More for fun, if I insert a 50 ms delay after each loop then all messages show as received.

jlpoltrack, 2025-03-18:
@festlv Are you testing only using this Py Script?  If so, there's something funny going on here - it seems to always emit a message of 180 bytes, the third one in sequence should be 20...

[image]

[image]

festlv, 2025-03-18:
> [@festlv](https://github.com/festlv) Few notes:
> 
> * Can you confirm you're using USB on the Tx?  (Based on the Py script I guess you are)

No, on TX I'm using the same UART that goes to bluetooth module (forgot whether it's 1 or 2), via USB-ttl adapter.

> * Have you tried using the Serial 2 / LPUART pins?  If not, could you?

I'll try tomorrow.

> * Are you testing only using this Py Script?
No, the problems were initially observed with MissionPlanner's RTK injection feature.

But I think what you're observing is due to: 
a) pymavlink being brain-dead when serializing the message (take a look at the gps_rtcm_data_encode)
b) pymavlink serializer not checking whether the trailing content of MAVLink message payload is 0s, and can be left out of the message (IIRC MAVLink spec allowed this).

jlpoltrack, 2025-03-18:
> No, the problems were initially observed with MissionPlanner's RTK injection feature.

Ah, good to know.  My general thought is that it would be best if we can confirm that MissionPlanner potentially does these things as well:
- Messages being longer than their specified length (I would think maybe not)
- Zero gap between messages (I would think maybe yes)

jlpoltrack, 2025-03-18:
@olliw42 Is there a missing break in the `parse_serial_in_link_out` function?  I added this line and now the Py Script works... 

[image]

jlpoltrack, 2025-03-18:
@festlv If you have STM32 Cube Programmer you can try out these firmwares with the above fix + USB RxBufSize is now 2048.

[mr24_firmwares.zip](https://github.com/user-attachments/files/19326574/mr24_firmwares.zip)

olliw42, 2025-03-18:
ohhhh ... this is indeed missing ... good catch !!

the other break are correctly located IMHO. They are inside the "higher" `if (fmav_parse_and_check_to_frame_buf(&result, buf_ser_in, &status_ser_in, c)) {` which is the correct thing to do. You caan check, this function returns true upon having parsed a complete message, either a "unknown" or a "known" one. 
WHat we should do is to have that if also in the other case where the break is missing.

jlpoltrack, 2025-03-18:
For completeness, the script as is generates a data rate of ~2,156 bytes / sec which is too much for 31 Hz (2 kBytes / sec) and will show missed messages.  Changing the data lengths to all 160 generates a data rate of ~1,914 bytes / sec which is 95% of capacity will show no missed messages.

olliw42, 2025-03-19:
> USB RxBufSize is now 2048

might be just a placebo, see comments to PR 💯

festlv, 2025-03-19:
@jlpoltrack, @olliw42  thank you for looking into and fixing this!

I'm in position to at least attempt building it myself (haven't used stm32cubeide in ages, setting up the development environment per mLRS-docu right now) so I'll try.

jlpoltrack, 2025-03-19:
@festlv In case it got lost, there are also some binaries here: https://github.com/olliw42/mLRS/issues/308#issuecomment-2734526994

festlv, 2025-03-19:
@jlpoltrack thanks, I just wanted to setup the development environment myself, as I also needed firmware for 900MHz mateksys devices eventually and did not want to bother you.

So far I can confirm that the issue is fixed on my desk setup, but I can't test the flying setup yet, as the NTRIP service I am using is down. I'll also report back if this change would fix MAVFTP failing ~ half of the time when connecting via MissionPlanner, as it sounds like it could be related.

jlpoltrack, 2025-03-19:
@festlv 

Sounds good.  Is it possible to confirm with your setup what the actual data requirements are for RTK?  The web is somewhat vague here giving rates from 500 Byte/s to 1.5 kBytes/s...

festlv, 2025-03-19:
> [@festlv](https://github.com/festlv)
> 
> Sounds good. Is it possible to confirm with your setup what the actual data requirements are for RTK with Mission Planner? The web is somewhat vague here giving rates from 500 Byte/s to 1.5 kBytes/s...

Well, the range you found on web is actually exactly what I am observing. 

The values I'm quoting come from MissionPlanner's RTK inject feature (output data rate, but not sure if it includes MAVLink framing overhead or not):
1) Using corrections from online service via NTRIP: ~1.5KiB/s
2) Using corrections from RTK-capable u-blox receiver: ~500byte/s.

The rate definitely depends on which constellations corrections are provided for- in my example the NTRIP service I'm using provides corrections for all four- GPS, Glonass, Beidou, Galileo, whereas u-blox didn't provide any corrections for Beidou)

However there's probably more to it that I can't explain (e.g. in the example above, I would expect u-blox data rate to be 3/4 of the NTRIP data rate, but it's actually 1/3- which is where I must confess that I lack deep understanding of the actual values in the RTCM data stream to give more insight on what could explain the difference.

jlpoltrack, 2025-03-19:
> The values I'm quoting come from MissionPlanner's RTK inject feature (output data rate, but not sure if it includes MAVLink framing overhead or not):

Assuming that it doesn't include framing, we could maybe guess with framing the worst case is 10% higher data rate, so 1.65 kBytes / sec - so should be possible on 31 Hz mode.  Would be good if you can confirm that it works across 31 Hz, 50 Hz and FLRC modes.

festlv, 2025-03-19:
My NTRIP service started working again, I can confirm that the current main branch fixes the original issue- MissionPlanner transmitted corrections from NTRIP are accepted by GPS receiver on UAV. 

After paying closer attention to the data rate reported by MissionPlanner, when MissionPlanner reports ~1500byte/s, in Wireshark traffic I observe closer to 1700byte/s, so MissionPlanner must be reporting RTCM payload length.

This still does fit within 31Hz mode, which is what I'm using on the 900MHz flying configuration.

olliw42, 2025-03-20:
the one thing I would be concerned about is that this RTCM thing appears to not allow any or not much loss of messages. 
It would mean that you need a very high LQ ... and we do not (yet) have retransmissions on the tx->rx side ...
that is, it may not work far in distance ...

festlv, 2025-03-20:
It is somewhat resilient, but only if the loss is random. GPS receiver can go without correction messages for some ten seconds or so before the fix type is degraded.

But in this case I'm fairly sure the messages were always transmitted in the same sequence, and it was always (roughly) the same types of messages that were lost.