olliw42/mLRS #311: Time stats broken on R9MX?

jlpoltrack, 2025-03-22:
On R9MX, Time stats seem to print out at 10x the rate you specify.

I added this line to mlrs-rx.cpp:

[image]

But this prints out a message every second instead of every 10 seconds, so it appears 10x faster than it should be.  Further the minimum time is incorrect and reported as 99 uS:

[image]

FYI: @brad112358 

Back of my mind wonders if this could be related those weird disconnects observed on R9MX.

olliw42, 2025-03-22:
didn't the R9MX had this thing with the oscillator ... is this maybe somehow related to that?
are there any other indications of incorrect timinings, or is it just that??
remind me, what weird disconnects?

jlpoltrack, 2025-03-22:
> didn't the R9MX had this thing with the oscillator ... is this maybe somehow related to that?

The oscillator was changed from the MSI to the HSI but when I look at the clock tree the APB1/2 timer clocks should be the same as the dividers look to be the same before and after this change.

> are there any other indications of incorrect timinings, or is it just that??

Not that I see, the SWUART and 1 Hz debug block work fine.

> remind me, what weird disconnects?

https://github.com/olliw42/mLRS/issues/221

brad112358, 2025-03-22:
The total count and number of calls is consistent with the rest of the reported values.  I don't see any explanation other than the timer is running 10 times faster than it is supposed to be running.  As to how that might be possible, I have no clue!

olliw42, 2025-03-22:
>>remind me, what weird disconnects?

>https://github.com/olliw42/mLRS/issues/221

I'm confused as regards the statement
didn't we identify the issue?
wasn't the TS_xxx code introduced exactly for diagnosing this and heavly used for this, and now it comes out wrong?

brad112358, 2025-03-22:
> I'm confused as regards the statement didn't we identify the issue? is the statement that it was in fact something else? wasn't the TS_xxx code introduced exactly for diagnosing this and heavly used for this, and now it comes out wrong?

Yes, we identified the issue, yes, I used an older version of this code extensively in debugging the issue.  I'm also very confused.

olliw42, 2025-03-23:
@jlpoltrack could you pl indicate all changes you made?
it seems to me that a TS_START(0) is needed, you do have that, right?
also, what is the t min which you expect to see? it's not the systick loop, i.e., it is called at max speed, isn't it?

jlpoltrack, 2025-03-23:
> [@jlpoltrack](https://github.com/jlpoltrack) could you pl indicate all changes you made?
> it seems to me that a TS_START(0) is needed, you do have that, right?

Only the `TS_END` call is needed and only need that one line in the above screenshot.

> also, what is the t min which you expect to see? it's not the systick loop, i.e., it is called at max speed, isn't it?

If the loop rate is measured as 86 kHz, then the min time should be something like 9 uS.

olliw42, 2025-03-24:
@jlpoltrack @brad112358 
ok, news, an issue certainly was that the R9MX hal files defined the same timer for both MICROS and SWUART. Effect was that micros was reconfigured to 10MHz insetad of the 1US, which made it run 10x faster.
This bug only affected the rx & tx R9MX, I have checked all other targets and they seem fine.
I wonder how the R9MX ever could work with debug enabled ...

anyway, with some test code I still don't get the output I expect ... let's see ...

closed 2025-03-24 by olliw42

brad112358, 2025-03-24:
Indeed, and I'm also wondering how I didn't notice this back when I first wrote the TS_ functions.  I could have sworn I used it on R9mx at the very end of our jitter fix testing.