olliw42/mLRS #321: Passthrough, Battery Capacity not available?

jlpoltrack, 2025-04-15:
Noticed that on mLRS the battery capacity is not populated in the Yaapu view, requiring one to use manual overrides:

[image]

Using ELRS does populate this correctly:

[image]

olliw42, 2025-04-15:
should be this pieceof code
https://github.com/olliw42/mLRS/blob/main/mLRS/CommonTx/crsf_interface_tx.h#L630-L640

has there maybe something changed in how ArduPilot does this message? Or this data coming form somewhere else?

I know that Yappuu has a pretty complicated config for batteries, maybe there's a difference here?

jlpoltrack, 2025-04-15:
I think not CRSF but Passthrough - so should be this mLRS code: https://github.com/olliw42/mLRS/blob/main/mLRS/Common/protocols/passthrough_protocol.h#L587-L606

jlpoltrack, 2025-04-15:
Last comment is mostly irrelevant :) I think Yaapu expects this as a parameter in the 0x5007 Passthrough Parameter Message:

https://github.com/yaapu/FrskyTelemetryScript/blob/abf55404e2de95e61795142447dfa9aecee23c70/OTX_ETX/c320x480/SD/WIDGETS/yaapu/main.lua#L1177-L1183

olliw42, 2025-04-15:
ah, that's the defined capacity of the battery not someething like consumption
yeah, this is not shown by mLRS. Reason is that one would have to add some stack to call the values of the respective parameters from the AP fc ... and to do that properly and reliably it would require quite a good bit of negotiation ... hence I didn't do it.
We do now have however also an autopilot class which does the work of finding and detecting the fc, so we could maybe do it now relatively cleanly. 
I'm not sure it's worth the effort however. If needed I think you can set that also in the yappuu config.

olliw42, 2025-04-15:
argh ... but that's rx side ... would have to do such a thing tx side ... or hope that whatever the rx does is good enough for the tx to get it ...

jlpoltrack, 2025-04-15:
> I'm not sure it's worth the effort however. If needed I think you can set that also in the yappuu config.

I defer to you on the effort (although if it is in the CRSF stream, could this be converted to Passthrough?) Yes, it can be set manually.  My note is here is that I use a single EdgeTx model for all of my mLRS gear, so I would have to change it whenever I switch models.

olliw42, 2025-04-15:
if it is not in passthur it can't be in the crsf stream ...

jlpoltrack, 2025-04-15:
> is that really such a relevant piece of info?? besides getting this display it doesn't do anything AFAIK

It works in conjunction with the capacity consumed to tell you what battery percentage you are at, so rather nice IMO.  I will probably add a note in the Docs saying that capacity must be manually applied.

olliw42, 2025-04-15:
but there is a battery precentage directly, isn't it (at least in battery_status there is ...)

olliw42, 2025-04-20:
@jlpoltrack 
was trying to do the idea of reconstructing the batt capacity from the available consumed and remaing data
is in principle possible, but won't work until the battery has been used for while
reason is that if the battery is close to full `(100 - percentage)` is pretty small (or even zero) and we won't get a good value.
So, we could get that estaimate only after a certain while.
I have no idea, would you think this is acceptable? Would this confuse Yappu?

the alternative is really to make the tx ask for the BATT_CAPACITY parameter
since it's through the same uart on the fc which the GCS is also talking too it may confuse AP a bit, as well as the GCS.

jlpoltrack, 2025-04-20:
> since it's through the same uart on the fc which the GCS is also talking too it may confuse AP a bit, as well as the GCS.

I suppose I would prefer this approach for the reasons you mention, could Sys IDs be used as to not route the request to the GCS?

olliw42, 2025-04-20:
I've pushed a branch https://github.com/olliw42/mLRS/tree/dev-battcapacity
tested it on bench without any GCS in the game
seems to work well for me
needs more practical testing and also in combination with a GCS etc. pp. and blabla

olliw42, 2025-04-20:
> I suppose I would prefer this approach for the reasons you mention, could Sys IDs be used as to not route the request to the GCS? 

PARAM_VALUE unfortunately doesn't have targets, so when send by AP it will go to everywhere ... this actually was be design at some point, so that all components can be aware of a param change e.g. 

I have no idea how AP behaves if it is in the process of sending a response to a PARAM_LIST_REQUEST or a mftp param request. 
I have no idea how a GCS would behave if it gets a PARAM_VALUE out of order. i.e., not as expected it is in a PARAM_LIST_REQUEST or a mftp param request. I would think (hope) that a GCS is made robust, so that's probably fine. Would not bet money on AP however.

jlpoltrack, 2025-04-20:
So the battery capacity comes through fine, but as soon as 1 mAh is consumed, I get this value in the Capa element which doesn't make sense (AP 4.7):

[image]

olliw42, 2025-04-20:
oh ... I haven't checked for that ... didn't recall it has such a field LOL ... let's see

olliw42, 2025-04-20:
ähm ... I actually don't understand this since I have only touched passthrough ,..

olliw42, 2025-04-20:
if I'm not mistaken there is no CRSF sensor for the battery capacity, only one for the consumed capacity (which I presume is shown as CAPA) ... is it possible that yaapu is doing some faking?
(the battery of may radio are charging so can't quickly test)

jlpoltrack, 2025-04-20:
Changed to AP 4.6 on the FC, same thing.  Flashed to commits prior to this, same thing.  Grr, this was working...

olliw42, 2025-04-20:
if you set BATT_CAPACITY to zero, it should fallback to not sending the respective CRSF passthru message to the radio

olliw42, 2025-04-20:
I'm actually massively confused now
there are respective native CRSF sensors as it seems
but CAPA is the consumed mAh
and we constrain to 8388607 ...  https://github.com/olliw42/mLRS/blob/main/mLRS/CommonTx/crsf_interface_tx.h#L635

jlpoltrack, 2025-04-20:
Okay my bad, the FC I have on the bench somehow was reporting a negative current with just USB and therefore the overflow was occurring because the consumption was -1 mAh.

olliw42, 2025-04-20:
LOL ... I've noticed the neg current but was assuming AP wouldn't count back haha

jlpoltrack, 2025-04-20:
Yea, this is a signed number in Battery Status... 

[image]

olliw42, 2025-04-20:
it is signed to allow for the -1 to indicate it's invalid .... AFAIK
(like some other)

olliw42, 2025-04-21:
@jlpoltrack 
another approach to the issue would be to do the request for BATT_CAPACITY on the Rx side and introduce a mLRS specific mavlink message carrying these info which the Tx could request from the Rx. Is more complicated, but it could have the advantage that the negotiation with the fc happens immediately when the Rx sees an Fc, which is more likely at a different time then than a GCS connects. It could also have an advantage in the longer run that also other data about the fc could be send to the Tx without the TX having to request it (like e.g. the fc firmware version).
Not sure it's worth the effort though.

jlpoltrack, 2025-05-09:
@olliw42 Did you find any issues with GCS? What are next steps here?

olliw42, 2025-05-09:
ähm, I haven't actually tried anything further
I guess I will leave what the next steps are to you guys to decide (assuming you guys test it pratically :))

jlpoltrack, 2025-05-10:
I tried out two GCS scenarios where this would interact with normal stream:
- Active connection to GCS, save mLRS parameters causing restart - no issues
- Active connection to GCS, repower the FC + Rx - no issues

I would add that am pretty sure in normal use for me there wouldn't be any interactions as I typically wait to see the elements start to populate on the radio before connecting to a GCS and at this point the battery capacity would've already been received from the FC.

Concluding - I don't see any issues that would prevent this from being included.

olliw42, 2025-05-10:
great. Thx.
have you noted the idea of doing it on the rx side and sending it to the tx via a mavlink? Would make it more robust, but is also more complicated. Thoughts on this?

olliw42, 2025-05-10:
@brad112358 thoughts on this PR? Any issue with it from your side?

jlpoltrack, 2025-05-10:
> have you noted the idea of doing it on the rx side and sending it to the tx via a mavlink? Would make it more robust, but is also more complicated. Thoughts on this?

I suppose I defer to your motivation on this as I can't see a lot of future use cases :)

olliw42, 2025-05-10:
yo ... it's hard to tell what will work best ... I guess nowadays most will use mFTP anyway for param download ... so it shouldn't interfere with a GCS either way
let's keep it as it is and see how it goes

and let's wait for Brad's comments

based on the CSRF discussion in discord I actually wonder if we shouldn't add the CRSF tx wire too ... that way we could read the CRSF from the FC and actually pass that data along over the air ... this would give all the data required for yappu with a much less bandwidth ... I once did that with modifying ArduPilot and it required only ca 700-800 bytes/s, compared to the ca 1900 I nedd currently ... so this would be also a way to save a good junk on bandwidth ... anyway, somewhat OT LOL

brad112358, 2025-05-12:
I'm sorry I didn't find the time to look at this bug sooner.  I still have not reviewed the code, nor have I tested the branch yet, but I think I follow the discussion and understand the problem now.  I have a couple of questions/thoughts I'd like to raise before I try some testing: 
I really think the current approach is probably more complicated than it's worth since it's just for display purposes.  Are we sure that we can't estimate the capacity on the fly based on the BATTERY_STATUS message accurately enough?  If we notice when the reported percentage goes down by one since the last report, for example, from 100 to 99, and calculate the capacity only at this time, based on how we know Ardupilot calculates it,I think we can estimate the capacity much more accurately than what you found when you tried it.  We should be able to get a fairly accurate number while the battery is still at 99% full.

brad112358, 2025-05-12:
In case it's not clear why estimating capacity only when the percentage changes makes our estimate much more accurate, think of it as a way to recover most of the information lost in the rounding of the percentage to an integer.

olliw42, 2025-05-13:
thx for your comments
no worries as regards looking at it, I was just interested to see if there is another opinion

concerning the estimates, I've tried it and while doing it became clear to me that this can't work ... unless one is willing to wait to get a reasonable estimate only late in the flight or accept other significant compromises
I can't see (set?) that your arguments invalidate the conclusion, catching it at exactly the flip may make it more accurate, but "more accurate" is not the same as "accurate". We don't do math but numerical numbers. At 1% any error is amplified by a factor of 100.
You may prove me wrong, but I think it is inevitably a vulnerable approach.

brad112358, 2025-05-13:
How accurate does it need to be? And how soon do you need it displayed accurately?  With the strategy I suggested, if battery status is sent at 1 hz and the battery lasts less than 1 hour, we should have only about 2% error after about 1 minute.

olliw42, 2025-05-13:
how do you come to that 2% accuracy estimate?

as regarsd the other questions I can't say, need to ask @jlpoltrack ... I mean I didn't miss it for the last couple years LOL

brad112358, 2025-05-13:
Once per second, the FC reports the battery state.  Lets consider %used instead of %remaining since the math is easier:
%used = 100 * used/capacity
The message provides an integer value so %reported = int(%used) = %used - %error where %error is the discarded fractional part.
If the battery lasts 1 hour and we are flying and using a steady amount of power, then, the % remaining will drop by 1% approximately every 36 seconds.  When it changes from 1 to 2 % used at about 36 seconds into the flight, we can use the previous report and estimate the difference between the reported percentage (1%) and the actual percentage as % error < 1/36.

If we knew the exact error, we could calculate the actual capacity as cap = 100*used/(%reported+%error), but we don't know the exact error, so we calculate est_cap = 100*used/%reported  When %reported = 1, cap = 100*used.

We can estimate an upper bound on the error ratio as est_cap/cap = 1/(37/36) = 36/37, or between 2 and 3% error.  If the battery life is shorter, we can estimate sooner or more accurately.  As time goes on, our estimate becomes more accurate.  When 3% of the battery is used, our estimate will be within 1%.

jlpoltrack, 2025-05-13:
> [@jlpoltrack](https://github.com/jlpoltrack) Is a gradually improving estimate within 2-3% after 30 seconds to a minute of flight time good enough?

I don't see any issues with the accuracy - note that Yaapu only reports to the nearest 0.1 Ah so not a lot of precision is required.

My counterpoints to the approach of calculating would be:
- Some time is needed to generate the estimate, perhaps not a huge deal but could give user false impression that something is broken when there's no battery capacity shown.
- Generally changes the logic from how it currently works, e.g. expect it to just reflect whatever it set in the batt_capacity parameter.

brad112358, 2025-05-14:
I didn't think it currently works at all in main.  Anyway, it's not my call, I just wanted to point out that there is an easy passive way to get a good estimate if you are willing to wait until the percentage changes.

olliw42, 2025-05-14:
I think we will not do the estimates, for the reasons meantioned
but nevertheless, as it's interesting

for briefness, I will write the equation as 
r = 100 * u / cap + e 
(r = reported in %, u = used in float, e is the error in the reported %)
If I understand you correctly, you suggest to measure r and u at two points (exactly then r flips) and to substract, that is
r1 = 100 u1 / cap + e1
r2 = 100 u2 / cap + e2
The difference is then
r2-r1 = 100 (u2-u1) / cap + (e2-e1)
We get cap as
cap = 100 (u2-u1) / (r2-r1) + (e2-e1) * cap / (r2-r1)
Since we do the next tick, we have r2 - r1 = 1, and thus

cap = 100 (u2-u1) + (e2-e1) * cap

Now, an upper bound for e1 and e2 is +-0.5. The upper bound for e2-e1 is thus +-1 (in statistical average it will be sqrt(2) the error for e2 or e1). So your error term is as large as your value.
The only way to get out of this is to have/assume some correleation between the errors. This indeed should be the fact, since the flip is always such that r is decreasing, and thus clearly balanced to one side. This will reduce the actual error quite a bit. But it's really some dodgy territory now. 
I continue to think that this isn't a very robust approach ... if one wants to get some reasonable number in reasonable time.

As it's fun, let's continue to brain storm. Not taking the average we would get

cap = 100 * u / r  + e * cap

the upper bound for e is again +-0.5, and the error term would be massive. But note, it will lower than for the difference. This shows a general "rule" that differences always amplify the errors in numerical math. So, no benefit of takling differences.

Also if one would be willing to assume a certain preference in the errors, the difference thing would be worse.
We could assume the error e.g. to always be close to 0.5 (or -0.5 dependeing on how we cross the flip), so we could write e as e = e0 + ee, where e0 is some constant and ee is now the much smaller error of the error. Note how it's getting dodgy now, we may think we want to use e0 = 0.5, but since e is bounded by 0.5 it would mean that ee has some strange behavior itself. Anyway, let's go for it. With differences we would get the estimate

cap = 100 (u2-u1) + (ee2-ee1) * cap

and the error term would indeed have improved. But it's really hard to believe it would be now few % (like your 2-3%), given that before it was rather something like 50% or more ... I doubt anyone woud understand the error of the error that well ...
For the normal approach we would get the estimate

cap = 100 * u / (1 - e0) r + ee / (1 - e0) * cap

The error again is not worse as for the difference.

olliw42, 2025-05-14:
@jlpoltrack 
pushed an updated version, works fine for me on the bench, would appreciate retesting

I concluded that one doesn't need to request BATT_MONITOR first (tried it and it made the thing really more complicated). So O added a counter so we request for only for 10 times. This prevents that if the param is not existing (because BATT_MONITOR is didabled) we request for ever.
Has the advantage also that if someone else (a gcs) asks for parameters, mLRS will now also stop.

jlpoltrack, 2025-05-14:
Seems to work fine on the latest, shows the battery capacity as expected.

Did you find 10 times necessary?  Would've thought a lower number would've been fine too.

olliw42, 2025-05-14:
merged

MANY THX folks

closing

closed 2025-05-14 by olliw42

brad112358, 2025-05-14:
I'm glad it's working well now.  No need to change working code.  But since you find it fun...

> If I understand you correctly, you suggest to measure r and u at two points (exactly then r flips) and to substract, that is

No.  We don't use the message after r flip.  Only the message before it flips which is when e is the smallest!

The current draw does not vary much unless we start/stop flying and the current used is always monotonically increasing.  If we see n reports before r flips, then e can be estimated as 1/n.  Much less than .5!  In my example, n was 36 while we are flying which results in e < 1/36 and an error in our estimate of < 3%.

Somehow, you have missed or are ignoring the entire point which is to take advantage of the fact that e is not really random.  It decreases until r flips, when it is relatively close to 0, then the sample after r flips, e becomes near 1 and starts decreasing again.

brad112358, 2025-05-14:
You could, instead use the message where e flips if you want to store less state, as long as you account for the fact that e is close to 1 at that point.