olliw42/mLRS #334: Bug: LOAD in mLRS-bw-luac script

vrquaeler, 2025-06-19:
Hi gentlemen, 
The LUA script for small screen radios (mLRS-bw-luac) seems to have a minor annoyance. 
The (re)_load_ button does not work on the 2nd and upcoming pages. The button works properly on the first page.

**Hardware**:
**RM Zorro** (OTX 2.3.15) with external modules:
- RM Bandit Nano
- DIY E28 F103CB
- Wio E5 Mini
 
**RM Pocket** internal ELRS ETX 2.10.6
- internal module flashed with mLRS v1.3.6 or latest dev build (v1.3.7)

**Steps to reproduce the effect:** 
- Enter script. 
- Go to second page or further. 
- Change a parameter. 
- Press "load". Do **not** press "_save_".

**Expected behavior:**
- The parameter should go back to it's initial value. 

**Discovered behavior**
- The parameter keeps its new value.

**other actions done:**
- checked the CLI for the external modules, "_reload_" works as expected
- run the external small bay modules on RM16S via BetaFPV module adapter, "reload" works as expected.

olliw42, 2025-06-19:
@brad112358

olliw42, 2025-06-19:
you mean in just the bw lua, right?
if so surely no objection, the bw lua is your baby 😂

olliw42, 2025-06-21:
closing this
I'm assuming the PR of brad did the tirck.
If not, pl do not hesitate to reopen :)

MANY thx for the report and your help in the matter.

vrquaeler, 2025-06-24:
Figured it out for myself. Optionally I can make a pr to give the "load" button a more meaningful  name.

olliw42, 2025-06-25:
reopened it.

Is there a way to get both sides into one code?

vrquaeler, 2025-06-25:
Hi, 
Yes, I can agree with @brad112358 's ideas. 
I was a bit uncertain about my PR. That's why I was asking for his opinion.

Nobody else seems to worry about that.

We may close this.

closed 2025-06-26 by olliw42