olliw42/mLRS #342: RP4-TD UART-bridge setup enters failsafe (100 ms red blink) and never reconnects

ddosoff, 2025-08-03:
Good day, OlliW 👋

First of all, thank you for your tremendous work on mLRS!
What I’m doing

    Two RadioMaster RP4-TD 2.4 GHz boards

        TX side – flashed with tx-radiomaster-rp4td-2400-sik-telem

        RX side – flashed with rx-radiomaster-rp4td-2400

    Using them purely as a UART bridge (simple “pinger” app on the serial ports confirms the link).

Issue

    Link works flawlessly at short range.

    When I walk far enough for the RX to lose the signal, it correctly enters failsafe and the red LED begins flashing very rapidly (~100 ms period).

    After that the RX never regains the link, even when I move back into range; only a rx power-cycle helps.

Expected behaviour

RX should automatically exit failsafe and reconnect to the TX once the signal returns.
Actual behaviour

    Remains in failsafe indefinitely.

    LED keeps flashing at ~10 Hz.

    No data are forwarded over UART until the RX is rebooted.

Environment

    mLRS commit: … <– (current master, cloned 2025-08-03)

Could you please point me to the proper fix or advise how to make the RX re-enter LISTEN → SYNC → CONNECTED automatically?

Thank you! 🙏

olliw42, 2025-08-04:
this does not look like a normal failsafe

it looks to me like it goes into one of the failure modes
you say it blinks at 10 Hz. Can you pleae describe in more detail the blink pattern of the leds. For both rx and tx (they can be different)

@jlpoltrack @brad112358 @tmcadam   I suspect an issue with that one of them goes into a FAIL. Given the past, a luikely reason could be on the tx side with some timing. Since I'm not so much on the esp side of things, your input would be valuable :)

jlpoltrack, 2025-08-05:
> But RX still reboots some time on weak signals. I see ESP booting message on the UART.

Can you try and capture the crash + boot message from the ESP? The crash message should be right before the boot message.

ddosoff, 2025-08-28:
I am in the UART bridge mode. No flight controller. Just ping + echo via UART.
To reproduce the weak signal you can do this:

[image]
[image]

No code modifications.

brad112358, 2026-10-16:
@ddosoff It has been quite some time since this issue has seen any activity and, of course, a number of code improvements and bug fixes have been merged in the last year.  It would be very helpful if you could leave a comment here and let us know if you noticed that the issue has been fixed or if you are still able to reproduce it with the latest 1.4.04 release.

Thanks!

ddosoff, 2026-10-20:
> requests on the go with GitH quickly before datub Mobile for iOS
> and Android
> Download it today!

Hi Brad,

Will answer quickly before detailed answer.
I am using mlrs bridge with RP4-TD long time and it's mostly good. But
sometimes long delay occur around 4 seconds.
I am going to investigate the reason this winter and will share here all
details.

BTW I'm going to add some protocol like AT commands above the bridge to be
able to get stats from the same UART.

Thanks for your care.

brad112358, 2026-10-21:
Thanks for your response!

I'm concluding from your comment that this issue does not appear in recent firmware versions.  The delay sounds like an unrelated problem.  If your investigation can reproduce it, please open a new issue when you have more information.

Thanks!

ddosoff, 2026-11-05:
Hi Brad and Olli, 

Finally I found and fixed the issue. I extended mLRS to report local radio diagnostics—ESP32 temperature, RSSI/SNR, link quality, TX power, etc.—through the same UART used for application traffic.

We recorded an INT_WDT reset on an RP4TD. The crash record does not establish the exact cause, but there is a potential SPI re-entry path: both SX1281 chips share one bus, and the RX DIO handlers call GetAndClearIrqStatus() and ReadBuffer() while the main loop also accesses the radios. The ESP32 SPI wrapper uses spiTransferBytesNL() without transaction-level protection.

Our fork already defers TX DIO handling to service_sx_dio() in the main loop. I suggest applying the same approach to RX: keep both DIO ISRs limited to signalling pending work, then perform all radio SPI operations in one execution context. Consume pending flags atomically and service them before processing the radio state machine, preserving RX timing.

Any additional tasks needing radio access should use that same owner. Avoid a blocking mutex inside the ISR: it could interrupt the task holding the lock and deadlock. Serialization must cover the entire CS-low → command/data → CS-high transaction.

The fork brakes original mLRS logic, that's why I don't made the patch. But you can check https://github.com/Jeto251/mLRS now it works without any issues.

olliw42, 2026-11-06:
@ddosoff 
many thx. Good work! We indeed have also noted cases which we diagnosed to be the very same problem you are reporting here. We are however still in an exploration of what best to do.
Your approach to move all into main was also considered. I don't like it so much however since the handling can then be delayed quite a bit since all the other stuff is being done. In the tx code we thus had to introduce relatively sophisticated time gating&slotting to code. This also would then be needed in the rx code. I don't like that (and it needs these new wrappers in the isr, not really making it simpler)
We do have two PRs with two other approaches:
- https://github.com/olliw42/mLRS/pull/545
- https://github.com/olliw42/mLRS/pull/555

My preference is the second PR, but we have some ongoing debate over which is better/more correct.
Maybe you have a chance to test the second, or even both.

> Avoid a blocking mutex inside the ISR: it could interrupt the task holding the lock and deadlock

my understanding of RTOS is limited, can you pl elaborate on that comment (it seems to me to rule out the 1st PR).

ddosoff, 2026-11-13:
Now it's winter time, we don't fly.
Will give you feedback next spring.

olliw42, 2026-11-14:
anyway, we will settle for this https://github.com/olliw42/mLRS/pull/556, which is along the lines you took.
so, I will close this. If there is need, can be reopened.
Many thx for the feedback.

closed 2026-11-14 by olliw42